intel/media-driver #824: [Encode][mb_rate_control] different behavior between VMEPAK and VDENC when setting to default

fulinjie, 2020-01-15:
Could be verified using ffmpeg vaapi:

ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -r:v 30 -f rawvideo -s:v 352x288 -pix_fmt yuv420p -i MOBILE_352x288_30_orig_01_352x288_I420.yuv -vf format=nv12,hwupload -c:v hevc_vaapi -profile:v main -low_power 1 -b:v 100k -maxrate 100k -vframes 300 -y vaapi_lp.h265

If setting mb_rate_control = 0(default) :

In VMEPAK, the mb_rate_control will be disabled in driver. 

In VDENC, the mb_rate_control will be enabled in driver.

This affects the encoding quality.

Xiaogangli-intel, 2020-09-09:
Please provide the platform information, thank you.

XinfengZhang, 2021-01-09:
@Xiaogang-Li , lets check it on TGL, if no such issue, we could close it

Xiaogangli-intel, 2021-09-22:
Close it, there's no such issue now.

closed 2021-09-22 by Xiaogangli-intel