intel/media-driver #825: [mb_bd_rate_control][ffmpeg][vaapi] get poor quality if mb_rate_control is enabled

fulinjie, 2020-01-15:
CMDline:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -r:v 30 -f rawvideo -s:v 352x288 -pix_fmt yuv420p -i MOBILE_352x288_30_orig_01_352x288_I420.yuv -vf format=nv12,hwupload -c:v hevc_vaapi -profile:v main -low_power 1 -b:v 100k -maxrate 100k -vframes 300 -y vaapi_lp.h265

The quality (bd-rate) become poor if we explicitly enable mb_rate_control.

Xiaogangli-intel, 2020-09-09:
Which platform are you working on?

XinfengZhang, 2021-01-09:
ideally , mb_rate_control will enable MB BRC , which will reduce bitstream size with same quality, but sometimes, it is already enabled by default inside driver.  because no platform mentioned in the issue description , @Xiaogangli-intel , could we try the command on TGL, and check the difference, if there are no such issue, we could close it. thanks

Xiaogangli-intel, 2021-09-22:
Close it, no activities for more than 9 months.

closed 2021-09-22 by Xiaogangli-intel