intel/media-driver #826: Build errors in Ubunut Disco

zhoujd, 2020-01-15:
Our CI found following build error:

Following build error: in https://media-ci.ostc.intel.com/build/370984 on Ubuntu Disco (19.04)

06:21:41,027 INFO  - -- The C compiler identification is GNU 8.3.0
06:21:41,161 INFO  - -- The CXX compiler identification is GNU 8.3.0

09:18:12,454 INFO  - In file included from /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/cm_device_os.cpp:24:
09:18:12,454 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h: In function ‘int parse_separate_sysfs_files(int, int, drmPciDeviceInfoPtr, bool)’:
09:18:12,455 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:555:62: error: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 231 [-Werror=format-truncation=]
09:18:12,455 INFO  - snprintf(device->driverInfo, sizeof(device->driverInfo), "Driver Module %s Bus %s IRQ %s", driver_name, dev_path, buffer);
09:18:12,455 INFO  - ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~
09:18:12,455 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:555:13: note: ‘snprintf’ output 26 or more bytes (assuming 1049) into a destination of size 256
09:18:12,455 INFO  - snprintf(device->driverInfo, sizeof(device->driverInfo), "Driver Module %s Bus %s IRQ %s", driver_name, dev_path, buffer);
09:18:12,455 INFO  - ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
09:18:12,455 INFO  - In file included from /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/cm_device_os.cpp:24:
09:18:12,455 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:599:34: error: ‘/’ directive output may be truncated writing 1 byte into a region of size between 0 and 4096 [-Werror=format-truncation=]
09:18:12,455 INFO  - snprintf(path, PATH_MAX, "%s/%s", pci_path, attrs[i]);
09:18:12,455 INFO  - ^~~~~~~
09:18:12,455 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:599:17: note: ‘snprintf’ output 2 or more bytes (assuming 4098) into a destination of size 4096
09:18:12,455 INFO  - snprintf(path, PATH_MAX, "%s/%s", pci_path, attrs[i]);
09:18:12,455 INFO  - ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
09:18:12,456 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h: In function ‘int parse_config_sysfs_file(int, int, drmPciDeviceInfoPtr)’:
09:18:12,456 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:628:30: error: ‘/config’ directive output may be truncated writing 7 bytes into a region of size between 0 and 4096 [-Werror=format-truncation=]
09:18:12,456 INFO  - snprintf(path, PATH_MAX, "%s/config", pci_path);
09:18:12,456 INFO  - ^~~~~~~~~~~
09:18:12,456 INFO  - /opt/media/build/intel-media-driver/cmrtlib/linux/hardware/drm_device.h:628:13: note: ‘snprintf’ output between 8 and 4104 bytes into a destination of size 4096
09:18:12,456 INFO  - snprintf(path, PATH_MAX, "%s/config", pci_path);
09:18:12,456 INFO  - ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
09:18:12,591 INFO  - cc1plus: all warnings being treated as errors
09:18:12,596 INFO  - make[2]: *** [cmrtlib/linux/CMakeFiles/igfxcmrt.dir/build.make:206: cmrtlib/linux/CMakeFiles/igfxcmrt.dir/hardware/cm_device_os.cpp.o] Error 1
09:18:12,596 INFO  - make[2]: Leaving directory '/opt/media/build/intel-media-driver/__cmake_build'
09:18:12,596 INFO  - make[1]: *** [CMakeFiles/Makefile2:109: cmrtlib/linux/CMakeFiles/igfxcmrt.dir/all] Error 2
09:18:12,596 INFO  - make[1]: *** Waiting for unfinished jobs....

Please fix it.

xhaihao, 2020-01-16:
I am experiencing the same build error, can anyone fix it ASAP?

dvrogozh, 2020-02-11:
Same for Eoan.

I believe issue should be finally fixed with https://github.com/intel/media-driver/commit/85458f81e66470f55cdb323109edfaa5689e612a. Can someone, please, double check on the latest master?

zhoujd, 2020-02-11:
It looks like the issue is still the there for latest master: intel-media-driver-2711a3a1c333
https://media-ci.ostc.intel.com/build/378050/log

add commit id 85458f8 failed as well.

zhoujd, 2020-02-12:
It looks like the build error is gone.
I believe the commit id: **ec2d498378cabdbc2e4cabc730855031ec84400b** may fix the issue.
Thanks for the fix.

zhoujd, 2020-02-14:
I am closing this issue.

closed 2020-02-14 by zhoujd