intel/media-driver #827: [HDR] metadata units don't match between VA-API definition and driver implementation

sunxinpeng, 2020-01-16:
The definition of HDR10 metadata in VA-API:
```
/**
     * \brief The maximum display luminance of the mastering display.
     *
     * The value is in units of 0.0001 candelas per square metre.
     */
    uint32_t    max_display_mastering_luminance;
    /**
     * \brief The minumum display luminance of the mastering display.
     *
     * The value is in units of 0.0001 candelas per square metre.
     */
    uint32_t    min_display_mastering_luminance;
    /**
     * \brief The maximum content light level.
     *
     * The value is in units of 0.0001 candelas per square metre.
     */
    uint16_t    max_content_light_level;
    /**
     * \brief The maximum picture average light level.
     *
     * The value is in units of 0.0001 candelas per square metre.
     */
    uint16_t    max_pic_average_light_level;
```
According to above definition, these four metadatas use 0.0001 candelas per square metre as the unit. However, it seems that iHD driver doesn't do any unit conversion and use 1 candelas per square metre as the unit for these four metadatas directly.

@FurongZhang Could you help to confirm this issue？

uartie, 2020-04-02:
I think the VAAPI documentation is wrong for `max_pic_average_light_level` (MaxFALL) and `max_content_light_level` (MaxCLL).  Those two fields should be in units of 1 candela per square meter (1 nit).

For example, MaxCLL 1000 nits (which is common) can't fit into a uint16_t if it's supposed to be converted to units of 0.0001 nits.

uartie, 2020-04-02:
https://github.com/intel/libva/pull/402

sunxinpeng, 2020-04-03:
> I think the VAAPI documentation is wrong for `max_pic_average_light_level` (MaxFALL) and `max_content_light_level` (MaxCLL). Those two fields should be in units of 1 candela per square meter (1 [...]

The units of 0.0001 nits is defined by the spec for HEVC video side data, so I think the VAAPI document is right but the code is wrong. IMHO, this issue should be fixed by changing the code instead of the document in libva and iHD driver.

uartie, 2020-04-03:
@sunxinpeng AFAICT, only `max_display_mastering_luminance` and `min_display_mastering_luminance` are defined as 0.0001 nit units, so the VAAPI docs are correct for those two fields.

But for `max_content_light_level` (MaxCLL) and `max_pic_average_light_level` (MaxFALL), the only information I found (on the web) was that these are in 1 nit unit.  In gstreamer code, they also use 1 nit unit for these latter fields and point to `CEA-861.3, Appendix A` as the spec for these... unfortunately, I don't have access to that CEA spec.  If you have that spec, please send it to me.

uartie, 2020-04-03:
https://www.mysterybox.us/blog/2016/10/19/hdr-video-part-3-hdr-video-terms-explained (see Metadata section)

FurongZhang, 2020-04-07:
max_display_mastering_luminance and min_display_mastering_luminance specify the nominal maximum and
minimum display luminance, respectively, of the mastering display in units of 0.0001 candelas per square metre.
min_display_mastering_luminance shall be less than max_display_mastering_luminance.

FurongZhang, 2020-04-07:
Yes, https://github.com/intel/libva/pull/402 correct the comment. MaxCLL and MaxFALL should be in in units of candelas per square metre. But  max_display_mastering_luminance and min_display_mastering_luminance are in units of 0.0001 candelas per square metre.

FurongZhang, 2020-04-07:
@uartie @sunxinpeng I have reviewed Libva PR and merged that, so I closed this issue, please let me know if anything else, thanks a lot in advance!

uartie, 2020-04-07:
@FurongZhang intel/libva#402 only addresses half the issue reported here.

According to the issue description, `max_display_mastering_luminance` and `min_display_mastering_luminance` are still being treated as 1 nit unit in the driver... so this should be a driver-side bug.  @sunxinpeng is that correct?  If so, please reopen.

FurongZhang, 2020-04-08:
@uartie  I will check driver and fix the issue if it is really an issue by next week.

uartie, 2020-04-10:
@FurongZhang please reopen this issue until that can be clarified.

FurongZhang, 2020-04-30:
https://github.com/intel/media-driver/pull/923

FurongZhang, 2020-04-30:
I filed PR #923 to fix this issue. Invite you to review the code, thanks a lot in advance! 
@uartie

FurongZhang, 2022-11-18:
Will check in the code.

closed 2022-11-18 by FurongZhang