intel/media-driver #829: vpp load balance issue for gstvaapi and ffmpeg-vaapi.

xuguangxin, 2020-01-19:
test platform: ICL
commit id:
```
commit 79634e7db3604d6a4226c32a13c7135c0aded0d0 (HEAD -> master, origin/master, origin/HEAD)
Author: ChenJasonK <Jason.k.chen@intel.com>
Date:   Mon Jan 6 19:06:15 2020 +0800

    [VP] Update Wa for 128/256 mode setting

    Update Wa for 128/256 mode setting for compression

    Change-Id: I7e627d2858a35bf52275d2d390788bd2db36e12a

```
steps:
1. run multi process decode + vpp like this
```
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
time gst-launch-1.0  -v filesrc location=/tmp/1080P.264 ! h264parse ! vaapih264dec ! vaapipostproc scale-method=1 ! "video/x-raw(memory:DMABuf),format=RGBA,width=300,height=300" ! fpsdisplaysink video-sink=fakesink sync=false &
[... cut ...]
#38 0x00007ffff72bde83 in gst_base_parse_loop (pad=<optimized out>) at ../libs/gst/base/gstbaseparse.c:3611
#39 0x00007ffff7f40ab9 in gst_task_func (task=0x5555556e3710) at ../gst/gsttask.c:328
#40 0x00007ffff7dbb854 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#41 0x00007ffff7dbb111 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#42 0x00007ffff7cc9669 in start_thread (arg=<optimized out>) at pthread_create.c:479
#43 0x00007ffff7bf1323 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

XinfengZhang, 2020-01-19:
thanks, will check it

xuguangxin, 2020-02-13:
@XinfengZhang , any update on this?
thanks

FurongZhang, 2020-02-13:
Is the issue caused by the below commit?
commit 79634e7db3604d6a4226c32a13c7135c0aded0d0 (HEAD -> master, origin/master, origin/HEAD)
Author: ChenJasonK <Jason.k.chen@intel.com>
Date:   Mon Jan 6 19:06:15 2020 +0800

    [VP] Update Wa for 128/256 mode setting

    Update Wa for 128/256 mode setting for compression

    Change-Id: I7e627d2858a35bf52275d2d390788bd2db36e12a

xuguangxin, 2020-02-13:
@FurongZhang , thanks for reply.
You mean after reverting the patch. the load balance is good?
Did you try my comandline?

thanks

XinfengZhang, 2020-02-13:
@FurongZhang  @xuguangxin , it should not related with the commit. we should pay attention to why the workload balance will be horrible after enable VP in the pipeline.

xuguangxin, 2020-02-13:
@FurongZhang yeah, I think I misleading you on this. Sorry for this
The commit id I provided just for reproducing the issue.