intel/media-driver #836: [regression] commit 41ae6fb1cfe456a1926480a7a505585df8fae468 results in regression when sharing a ARGB surface with the 3rd party library

xhaihao, 2020-02-11:
The display image is corrupted when running the command below on ICL:

$> GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! video/x-raw\(memory:DMABuf\),format=BGRA ! glimagesink

it works well after reverting commit 41ae6fb1cfe456a1926480a7a505585df8fae468

VUNNYSODHI, 2020-02-13:
Hi @xhaihao 
I don't have https://github.com/intel/media-driver/commit/41ae6fb1cfe456a1926480a7a505585df8fae468 in my media-driver tree as am using https://github.com/intel/media-driver/commit/12b7fcded6c74377ecf57eb8258f5e3d55ca722e (tag/intel-media-19.4.0r) , still i see green screen issue 
[image]
  with NV12 format
**#GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! 'video/x-raw(memory:DMABuf),format=NV12' ! glimagesink**

XinfengZhang, 2020-02-13:
@FurongZhang @Jister

VUNNYSODHI, 2020-02-14:
Hi @FurongZhang @Jister @xhaihao 
Shall I raise other ticket to track for NV12 format? or continuing in this ticket is okay?

xhaihao, 2020-02-14:
@VUNNYSODHI The issue you saw should be different from this issue.  I can't reproduce your issue with tag/intel-media-19.4.0r on ICL.

VUNNYSODHI, 2020-02-14:
Thanks @xhaihao for the info, I have created separate ticket to track it https://github.com/intel/media-driver/issues/843

HeJunyan, 2020-02-15:
hi @FurongZhang and @xhaihao ,
And on SKL, I also check the pipeline:
gst-launch-1.0 filesrc location=AUD_MW_E.264 ! h264parse ! vaapih264dec ! glimagesink
get garbage output. when use EGL and DMABuf.

I root caused that the VABufferInfo->mem_size of that DMA buffer is not correct, less than the correct size.

xhaihao, 2020-02-18:
The bad commit has been reverted from master branch.

commit 9e18e315430f2b419bedfe7b8ce89c48881a8169
Author: Chen_JasonK <jason.k.chen@intel.com>
Date:   Thu Feb 6 16:08:45 2020 +0800

    Revert "[VP] Add MC modifier support in vaExportSurfaceHandle"

closed 2020-02-18 by xhaihao