intel/media-driver #841: Please check the caps for Gen12.

xuguangxin, 2020-02-12:
According to https://github.com/intel/media-driver/pull/840. There are many Gen12 caps mismatched with other platforms. The patch only fixed hevc, I believe vp9 10 bits has a similar problem too.
@XinfengZhang  Could you help review all caps for Gen12? Make sure it matches the hw capabilities.

Also, I saw a huge duplicate code in MediaLibvaCapsG12. could we refine it with MediaLibvaCaps?

thanks

edwarddavidbaker, 2020-02-26:
Hi, thanks for starting this issue. On the topic of 10 bit VP9, some of the testing frameworks for Chrome OS use a small binary for checking hardware media capabilities. This utility returns false for accelerated VP9 profile 2 decode since VA_RT_FORMAT_YUV420_10BPP is not advertised. 
```cpp
/* Helper function for detect_video_acc_vp9_2.
 * Determine given |fd| is a VAAPI device supports VP9 decoding Profile 2, i.e.
 * it supports VP9 profile 2, has decoding entry point, and can output YUV420
 * 10BPP format.
 */
static bool is_vaapi_dec_vp9_2_device(int fd) {
#if VA_CHECK_VERSION(0, 38, 1)
  VAProfile va_profiles[] = {
    VAProfileVP9Profile2,
    VAProfileNone
  };
  if (is_vaapi_support_formats(fd, va_profiles, VAEntrypointVLD,
        VA_RT_FORMAT_YUV420_10BPP))
    return true;
#endif
  return false;
}
```

```
localhost ~ # vainfo -a
libva info: VA-API version 1.6.0
libva info: Trying to open /usr/lib64/va/drivers/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.6 (libva 2.6.0)
vainfo: Driver version: Intel iHD driver - 19.4.0
vainfo: Supported config attributes per profile/entrypoint pair
<snip>
VAProfileVP9Profile2/VAEntrypointVLD
    VAConfigAttribRTFormat                 : VA_RT_FORMAT_YUV420
    VAConfigAttribDecSliceMode             : VA_DEC_SLICE_MODE_NORMAL
                                             VA_DEC_SLICE_MODE_BASE
    VAConfigAttribDecProcessing            : VA_DEC_PROCESSING
    VAConfigAttribMaxPictureWidth          : 16384
    VAConfigAttribMaxPictureHeight         : 16384
    VAConfigAttribProcessingRate           : VA_PROCESSING_RATE_DECODE
<snip>
```

This is on a Gen12 system running libva-2.6.0, libva-utils-2.6.0, gmmlib-19.4.1, and libva-intel-media-driver-19.4.0. Thanks for the help!

xuguangxin, 2020-03-06:
@edwarddavidbaker yes, GStreamer vaapi will do such test too. 
This why all gstreamer vaapi 10 bits decoder cases will fail on TGL .
:)

Xiaogangli-intel, 2020-03-24:
Thank you, working on this

XinfengZhang, 2020-05-27:
issue should be fixed by 40955ae4f440eb81e7efe8c36727f08fb12a625d , but suppose it is not enough , because we should add profile to de-couple the RT format for different profiles https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.h#L46

PR #962 (2020-06-09): fix decode caps related issue
including vp9 10bit and hevc screen-extended-main-444

Fixes #893
Fixes #841
commit ca71e10f (2020-06-09): [Decode] Add corresponding RT format according to codec profile

Add more profile RT format support
Supported profiles include hevc screen-extended-main-444 and VP9 profile2.

Fixes #893
Fixes #841
closed 2020-06-09 by intel-mediadev