intel/media-driver #843: The display image is green when running the command below on TGL

VUNNYSODHI, 2020-02-14:
ENV:

- Media SDK and Samples: https://github.com/Intel-Media-SDK/MediaSDK/releases/tag/intel-mediasdk-19.4.0
- Driver: https://github.com/intel/media-driver/releases/tag/intel-media-19.4.0r
- Gmmlib: https://github.com/intel/gmmlib/releases/tag/intel-gmmlib-19.4.1
- libva: https://github.com/intel/libva/releases/tag/2.6.0

The display image is green when ran:

$GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! 'video/x-raw(memory:DMABuf),format=NV12' ! glimagesink

Output:
[image]

xhaihao, 2020-02-16:
I can reproduce this issue against the latest media driver (commit ba5db01) by using sample_vpp from https://github.com/Intel-Media-SDK/MediaSDK/ (commit 7802ba5) @XinfengZhang @FurongZhang

xhaihao, 2020-02-16:
sample_vpp command:
$> sample_vpp -vaapi  -scc yuy2 -sw 1920 -sh 1080 -dcc nv12 -i input.yuv -o output.yuv

FurongZhang, 2020-02-18:
@xhaihao  Do you know if this is a media driver regression?

VUNNYSODHI, 2020-02-19:
Hi @FurongZhang, 
Its not regression, using "GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! 'video/x-raw(memory:DMABuf),format=NV12' ! glimagesink" 
Corruption is seen on ICL platform also

Jister, 2020-02-21:
Hi @VUNNYSODHI , could you help to dump VP HLT log and VP input/output surface according to https://github.com/intel/media-driver/wiki/Video-Processing-Debug-Tool ? Thanks very much.

VUNNYSODHI, 2020-02-24:
Hi @Jister,
Attached dump on running with **GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc num-buffers=10 ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! 'video/x-raw(memory:DMABuf),format=NV12' ! glimagesink**
HLT logs [HLT-logs.txt](https://github.com/intel/media-driver/files/4244044/HLT-logs.txt) 
and VP dump  [vp_dump.tar.gz](https://github.com/intel/media-driver/files/4244045/vp_dump.tar.gz)

Jister, 2020-02-24:
Hi @VUNNYSODHI , thanks for your dumps. As I check the driver dumps, the driver output looks correct, not show green as you described. So i think media driver should have no issue, I suppose it might be a midware issue.

VUNNYSODHI, 2020-02-25:
Hi @Jister, if media driver dumps looks ok, then can we say gstreamer-vaapi parameters also correct?
Can we suspect on msdkvpp for Green color what shown on output surface?

Jister, 2020-02-25:
Hi, @VUNNYSODHI , yes, suppose to be a msdkvpp issue.

VUNNYSODHI, 2020-02-25:
Hi @Jister,
Green corruption seen with "sample_vpp -vaapi -scc yuy2 -sw 1920 -sh 1080 -dcc nv12 -i input.yuv -o output.yuv"

xhaihao, 2020-02-28:
@Jister @FurongZhang @VUNNYSODHI @XinfengZhang 

It seems there are two different issues:

* The output of VPP is wrong, which can be reproduced by `sample_vpp -vaapi -scc yuy2 -sw 1920 -sh 1080 -dcc nv12 -i input.yuv -o output.yuv`.  With latest master HEAD (commit 7857d6bea214b755d610c1a478d4e1e23da31cdb), the output quality is better than before but I still see a black line at the bottom. In the past (commit ba5db0187f6017c9ed029b5fa5e257ff2f2bb045), I saw corrupted output.  I don't see the black line on ICL. 

* A video surface can not be shared between media driver and iris driver on TGL.  You will see an assertion failure with debug version of iris driver when running the command below:

` $> GST_GL_PLATFORM=egl gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12,width=1920,height=1080 ! msdkvpp ! 'video/x-raw(memory:DMABuf),format=NV12' ! glimagesink ` 

> gst-launch-1.0: ../../mesa/src/gallium/drivers/iris/iris_resource.c:1030: iris_resource_from_handle: Assertion `isl_surf_created_successfully' failed.

This assertion failure indicates iris failed to create a surface from the handle shared from media driver. 

You will see green display image with release version of iris driver when running the above command on TGL. (assert() is ignored in release version, so the command can run)

This command works well on ICL even if I use iris driver for ICL.

VUNNYSODHI, 2020-02-28:
Thanks @xhaihao for the update, even on my side with debug version of mesa i see same assertion error with gstreamer command and it fails to render anything.
**gst-launch-1.0: ../src/gallium/drivers/iris/iris_resource.c:1030: iris_resource_from_handle: Assertion `isl_surf_created_successfully' failed.
Aborted (core dumped)**
Could it be GEN12 iris specific issue for surface creation?

xhaihao, 2020-02-28:
@VUNNYSODHI,  for the 2nd issue, I think it is a GEN12 specific issue for buffer sharing between media driver and iris driver.

Jister, 2020-03-02:
@VUNNYSODHI can you help to debug into mesa driver to find the reason why surface creation failed? I also tried to reproduce the issue, i can see the corrupted output, but not green output, i synced with @xhaihao ,  seems  DMAbuf doesn't take effect,.
[image]

Jister, 2020-03-10:
@VUNNYSODHI @xhaihao I submit a PR for black line issue, please try #871

xhaihao, 2020-03-17:
@Jister #871 fixed the black line issue.

commit d9b5c612 (2020-03-24): [VP] Fix #843

SFC can support bottom color-fill on Gen12, use SFC to handle such scenario if no scaling needed when set ScalingPreference as COMP.
commit 645e6b9b (2020-03-27): Revert "[VP] Fix #843"

This reverts commit d9b5c612a5cbe0efc41a8a6413dab44f89d1f190.
Change-Id: I9b425f2c46b5a4761ce881da4257efe2c75b37b1
closed 2020-03-24 by intel-mediadev