intel/media-driver #849: [HDR][H2S] The input metadata doesn't work when doing tone-mapping

sunxinpeng, 2020-02-17:
Whatever the metadata of input stream I change, the tone-mapping pipeline(HDR to SDR) always get the same ouput file(same md5 value).

FurongZhang, 2020-04-07:
I am checking that.

FurongZhang, 2020-04-30:
@sunxinpeng It is expected. The final output of H2S is SDR content. Hence, for all the SDR output, they should be same regardless of Input Image Luminance!

sunxinpeng, 2020-04-30:
> @sunxinpeng It is expected. The final output of H2S is SDR content. Hence, for all the SDR output, they should be same regardless of Input Image Luminance!

@xhaihao Hi, haihao, could you help to check with this?

FurongZhang, 2020-05-08:
The behavior is expected, hence I close the issue. Please reopen and comment it if any concern. Thanks a lot in advance!

closed 2020-05-08 by FurongZhang