intel/media-driver #854: Need ability to disable IPC usage for platforms beyond Android

ykku16github, 2020-02-20:
The following 2 files disable IPC usage for the Android platform:
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/os/mos_context_specific.cpp
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/os/mos_os_specific.c

There is a need to do the same thing on other platforms, such as Chrome OS.  

It would be nice if the IPC-related code is encapsulated by a slightly more meaningful name.  For example, the name could describe the feature being supported.  Chrome OS or any other platform can then switch it on or off when calling cmake.  This suggestion is credited to Miguel Casas <mcasas@chromium.org>.

XinfengZhang, 2020-02-21:
good idea, the IPC is used for gen9 task balance, we could use random dispatch to replace it.

paulmenzel, 2021-09-07:
What is the status of this? [It’s also relevant for using the Intel driver with Mozilla Firefox.](https://bugzilla.mozilla.org/show_bug.cgi?id=1619585#c46)

Darkspirit, 2021-09-08:
Regarding Firefox it should be noted that - still experimental - VAAPI decoding will be moved from the sandboxed content process to the unsandboxed GPU process. Current sandbox violations don't seem to matter then.

XinfengZhang, 2021-10-15:
actually, for Iris Xe, we do not need IPC any more, the IPC call is only for Gen9 (SKL, KBL,...) which has no virtual engine support. 

We already have a patch which remove the IPC call , but it is not merged because of regressions and concerns. 
for gen9 platforms like SKL.   it has two VDBoxes, so, to balance the task from different process, we need IPC call to reserve the task information on these two VDBoxes, then choose the "less busy" VDBox to dispatch new task. 
after change the policy to random, it will be "balanced" if the process number is enough, but if there are just 2 or 3 process, it is possible that all task will be dispatched to 1 VDbox...

ravenexp, 2021-10-15:
CPU: Core i5-1135G7
DEV ID: `0x9a49`

paulmenzel, 2021-11-08:
@XinfengZhang, thank you for working on this issue. As this issues seems to [prevent the Firefox developers to enable VA-API video decoding by default in Mozilla Firefox](https://bugzilla.mozilla.org/show_bug.cgi?id=1698778), do you have a time frame, when all generations will be fixed?

heftig, 2021-11-08:
It seems ICL is not fixed yet, either.

XinfengZhang, 2021-12-02:
#1293

eero-t, 2023-07-04:
> #1293 could be used to fix ICL, SKL...

@XinfengZhang Are the other platforms fixed by now i.e. could this be closed?