intel/media-driver #859: iHD_drv_vide library is big

lsandov1, 2020-02-27:
the `iHD_drv_video.so` library is about 160Mb in size which is big (IMO) for a relocatable library. Above is the way I am building it:

```
MEDIA_DRIVER_VER=intel-media-19.4.0r
MEDIA_DRIVER_REPO=https://github.com/intel/media-driver/archive/${MEDIA_DRIVER_VER}.tar.gz

wget -O - ${MEDIA_DRIVER_REPO} | tar xz && mv media-driver-${MEDIA_DRIVER_VER} media-driver && \
  mkdir -p media-driver/build && \
  cd media-driver/build && \
  export PKG_CONFIG_PATH="/usr/local/lib64/pkgconfig" && \
  cmake -DBUILD_TYPE=release -DBUILD_ALONG_WITH_CMRTLIB=1 \
        -DMEDIA_VERSION="2.0.0" \
        -DBS_DIR_GMMLIB=/home/gmmlib/Source/GmmLib \
        -DBS_DIR_COMMON=/home/gmmlib/Source/Common \
        -DBS_DIR_INC=/home/gmmlib/Source/inc \
        -DBS_DIR_MEDIA=/home/media-driver -Wno-dev \
        -DCMAKE_INSTALL_PREFIX=/usr/local .. && \
  make -j $(nproc) && \
  make install
```

lsandov1, 2020-02-27:
Including `ls` sizes

```
$ ls -la /usr/local/lib64/dri/iHD_drv_video.so
-rwxr-xr-x 1 root root 167035512 Feb 19 18:12 /usr/local/lib64/dri/iHD_drv_video.so
$ ls -lah /usr/local/lib64/dri/iHD_drv_video.so
-rwxr-xr-x 1 root root 160M Feb 19 18:12 /usr/local/lib64/dri/iHD_drv_video.so
```

XinfengZhang, 2020-03-02:
if you want, you could customize it , choose different supported platform and features
there are switch inside media_gen_flags.cmake & media_feature_flags.cmake

eero-t, 2020-03-03:
You can strip debug symbols ("strip -s <file>") to decrease library to 1/5th size.  Dynamic linker doesn't load debug symbol sections to RAM, only debuggers use them. Debug symbols are included to builds so that distributions can split them to a separate package when building the sources.

If you want less debug symbols, compilers have options to control that.

=> INVALID?

ceyusa, 2020-03-03:
https://twitter.com/ceyusa/status/1182309314561794049 :)

lsandov1, 2020-03-03:
thanks @eero-t @XinfengZhang @ceyusa. we are containering this media-driver so the less the better. I will look at cmake & compiler & strip options to make it smaller and let you know my result.

aaronovz1, 2020-03-11:
With debug symbols stripped the library is still 33Mb.

XinfengZhang, 2020-03-11:
is there expected size?

eero-t, 2020-03-11:
If you care more about size than speed, you can use CMake target / compiler option for that. It's hard to say  how much can that save, but I think at best it could be tens of percents.

And even if you care more about performance than size, it's still better to measure which option gives that.   Sometimes e.g.  -Os can provide better performance than -O3 if perf-critical parts of code happen to fit into smaller cache level (on your particular HW) when they're size-optimized (-Os) instead of there being large amount of inlining (-O3).

aaronovz1, 2020-03-11:
Definitely prefer speed, but on an embedded system size matters as well. I suspect the next step would be the ability to remove code for unnecessary platforms.

XinfengZhang, 2020-03-12:
yes, https://github.com/intel/media-driver/blob/master/media_driver/cmake/linux/media_gen_flags_linux.cmake but there are too many combination, TBH, I am not sure whether all the option could work fine.

eero-t, 2020-03-12:
I guess is that sub-options for specific platforms on given GPU generation are about their identification and setting correct config for them and shouldn't affect size that much, the actual functionality implementation differences are between GPU generations.

lsandov1, 2020-03-23:
With the above comments, I believe this issue can be close. BTW, striping all versus stripping just debug symbols yields basically the same library size, with just 1Mb less for the latter. I believe this is expected.

closed 2020-03-23 by lsandov1