intel/media-driver #860: VPP fails to convert to/from RGB color primaries

uartie, 2020-02-27:
Observed with 4K resolution, when doing VPP CSC between YUV and RGB color spaces and setting the associated VPP _color standard_ properties.  When the color **primaries are sRGB for one surface** (input or output) and the color **primaries are not sRGB for the other surface**, the VPP output results in a pure black surface.

This issue was first discovered and reported in gstreamer-vaapi here: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/issues/238.  This initial report indicates that this does not happen with 1080p resolution.  But it's possible other resolutions (including 4K) exhibit this issue.

We are able to work around this issue in gstreamer-vaapi (https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/merge_requests/287) by always forcing the **output** color **primaries** to be the **same** as the **input** color **primaries**.

This appears to be a platform agnostic issue (i.e. observed on SKL, KBL, ICL and TGL).

FurongZhang, 2020-03-05:
OK, I will check that as soon as possible.

FurongZhang, 2020-04-23:
For this issue, we have already fixed it in iHD driver. Please use the latest iHD driver to verify that. @uartie

FurongZhang, 2020-04-23:
@uartie If no more object, I will close this issue. Thanks a lot in advance!

uartie, 2020-04-23:
Can you point me to which commit fixes this?

uartie, 2020-04-23:
Found it: a39fe9bc051a8c3efa8f35122a1585981ec7f816

I tested it by reverting our WA in gstreamer-vaapi and the issue is gone.  Thanks!

closed 2020-04-23 by uartie

FurongZhang, 2020-04-24:
Thank you so much @uartie