intel/media-driver #861: tone-mapping doesn't work well on ICL

sunxinpeng, 2020-02-28:
The tone-mapping pipeline can not work well on ICL.

From gstreamer-vaapi side, when performing tone mapping to sRGB output color standard, the VPP pipeline fails.  For bt709 or bt2020-10 output color standard, the YUV channels are all messed up (greenish tone).

From ffmpeg-vaapi side, when performing tone mapping on ICL, driver will return error message like: "Illegal Hdr cases on unsupported m_Platform, turn off HDR." and the pipeline fails, too. The related code in driver is as following:
```
if ((pRenderPassData->bHdrNeeded) && (!MEDIA_IS_SKU(m_pSkuTable, FtrHDR)))
    {
        eStatus = MOS_STATUS_SUCCESS;
        VPHAL_RENDER_ASSERTMESSAGE("Illegal Hdr cases on unsupported m_Platform, turn off HDR.");
        pRenderPassData->bHdrNeeded = false;
}
```
MEDIA_IS_SKU returns false on my ICL but returns true on my CFL.

FurongZhang, 2020-03-05:
I will check that as soon as possible.

uartie, 2021-02-09:
@FurongZhang Ping.  Any update?

FurongZhang, 2022-11-18:
Fixed.

closed 2022-11-18 by FurongZhang