intel/media-driver #862: Crash when create/destroy image in gstreamer-vaapi.

HeJunyan, 2020-03-02:
pipeline:
gst-launch-1.0  videotestsrc num-buffers=1000 ! vaapih264enc ! h264parse ! vaapih264dec ! fakesink

will crash at 
```
#0  0x00007ffff6f53428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
intel/libva#1  0x00007ffff6f5502a in __GI_abort () at abort.c:89
intel/libva#2  0x00007ffff6f957ea in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7ffff70aee98 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
intel/libva#3  0x00007ffff6f9e37a in malloc_printerr (ar_ptr=<optimized out>, ptr=<optimized out>, str=0x7ffff70aef38 "free(): corrupted unsorted chunks", action=3) at malloc.c:5006
intel/libva#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:3867
intel/libva#5  0x00007ffff6fa253c in __GI___libc_free (mem=<optimized out>) at malloc.c:2968
intel/libva#6  0x00007fffed5b04ee in GmmMemAllocator::operator delete (ptr=0x7fffd874abd0) at /home/robinhe/media/gmmlib/Source/GmmLib/inc/External/Common/GmmMemAllocator.hpp:52
intel/libva#7  0x00007fffed6455dc in GmmLib::GmmResourceInfoLin::~GmmResourceInfoLin (this=0x7fffd874abd0, __in_chrg=<optimized out>)
    at /home/robinhe/media/gmmlib/Source/GmmLib/inc/External/Common/../Linux/GmmResourceInfoLin.h:53
intel/libva#8  0x00007fffed679fa9 in GmmLib::GmmClientContext::DestroyResInfoObject (this=0x7b9510, pResInfo=0x7fffd874abd0) at /home/robinhe/media/gmmlib/Source/GmmLib/GlobalInfo/GmmClientContext.cpp:466
intel/libva#9  0x00007fffee93c3f2 in DdiMediaUtil_FreeSurface (surface=0x7fffd874ab10) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:998
intel/libva#10 0x00007fffee912e2c in DdiMedia_DestroySurfaces (ctx=0x7e9200, surfaces=0x7fffe77fd1c4, num_surfaces=1) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva.cpp:2170
intel/libva#11 0x00007fffee91c8e0 in DdiMedia_GetImage (ctx=0x7e9200, surface=21, x=0, y=0, width=64, height=64, image=1) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva.cpp:4884
intel/libva#12 0x00007ffff49814e3 in vaGetImage (dpy=0x822960, surface=21, x=x@entry=0, y=y@entry=0, width=width@entry=64, height=height@entry=64, image=1) at ../va/va.c:1790
intel/libva#13 0x00007ffff5036a23 in gst_vaapi_surface_get_image (surface=0x7fffd800c310, image=image@entry=0x7fffe844ba40) at ../gst-libs/gst/vaapi/gstvaapisurface.c:714
intel/libva#14 0x00007ffff4ff7d4d in extract_allowed_surface_formats (direction=GST_PAD_SRC, specified_format=GST_VIDEO_FORMAT_NV12, img_formats=0x7fffe80062f0, display=0x72f100)
    at ../gst/vaapi/gstvaapipluginbase.c:1457
intel/libva#15 ensure_allowed_raw_caps (plugin=plugin@entry=0x8329b0, format=GST_VIDEO_FORMAT_NV12, direction=direction@entry=GST_PAD_SRC) at ../gst/vaapi/gstvaapipluginbase.c:1502
intel/libva#16 0x00007ffff4ff97dd in ensure_allowed_raw_caps (direction=GST_PAD_SRC, format=<optimized out>, plugin=0x8329b0, plugin@entry=0x0) at ../gst/vaapi/gstvaapipluginbase.c:1556
intel/libva#17 gst_vaapi_plugin_base_get_allowed_srcpad_raw_caps (plugin=plugin@entry=0x8329b0, format=<optimized out>) at ../gst/vaapi/gstvaapipluginbase.c:1553
intel/libva#18 0x00007ffff4ff42d0 in gst_vaapidecode_ensure_allowed_srcpad_caps (decode=0x8329b0) at ../gst/vaapi/gstvaapidecode.c:228
intel/libva#19 gst_vaapidecode_get_allowed_srcpad_caps (decode=0x8329b0) at ../gst/vaapi/gstvaapidecode.c:252
intel/libva#20 gst_vaapidecode_update_src_caps (decode=0x8329b0) at ../gst/vaapi/gstvaapidecode.c:280
intel/libva#21 gst_vaapidecode_negotiate (decode=0x8329b0) at ../gst/vaapi/gstvaapidecode.c:476
intel/libva#22 gst_vaapidecode_push_decoded_frame (out_frame=0x7fffe8450020, vdec=0x8329b0) at ../gst/vaapi/gstvaapidecode.c:536
intel/libva#23 gst_vaapidecode_push_all_decoded_frames (decode=0x8329b0) at ../gst/vaapi/gstvaapidecode.c:685
```

with patch
```
diff --git a/gst/vaapi/gstvaapipluginbase.c b/gst/vaapi/gstvaapipluginbase.c
index e0c1189..dc56285 100644
--- a/gst/vaapi/gstvaapipluginbase.c
+++ b/gst/vaapi/gstvaapipluginbase.c
@@ -1433,6 +1433,9 @@ extract_allowed_surface_formats (GstVaapiDisplay * display,
     if (img_format == GST_VIDEO_FORMAT_UNKNOWN)
       continue;

+    if (img_format != GST_VIDEO_FORMAT_RGBA)
+      continue;
+
     surface_format =
         (specified_format == GST_VIDEO_FORMAT_UNKNOWN) ?
         img_format : specified_format;

```

it crash at 
```
#0  0x00007ffff6f53428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
intel/libva#1  0x00007ffff6f5502a in __GI_abort () at abort.c:89
intel/libva#2  0x00007ffff6f957ea in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7ffff70aee98 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
intel/libva#3  0x00007ffff6f9e37a in malloc_printerr (ar_ptr=<optimized out>, ptr=<optimized out>, str=0x7ffff70aef38 "free(): corrupted unsorted chunks", action=3) at malloc.c:5006
intel/libva#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:3867
intel/libva#5  0x00007ffff6fa253c in __GI___libc_free (mem=<optimized out>) at malloc.c:2968
intel/libva#6  0x00007fffed5b04ee in GmmMemAllocator::operator delete (ptr=0x7fffd874e580) at /home/robinhe/media/gmmlib/Source/GmmLib/inc/External/Common/GmmMemAllocator.hpp:52
intel/libva#7  0x00007fffed6455dc in GmmLib::GmmResourceInfoLin::~GmmResourceInfoLin (this=0x7fffd874e580, __in_chrg=<optimized out>)
    at /home/robinhe/media/gmmlib/Source/GmmLib/inc/External/Common/../Linux/GmmResourceInfoLin.h:53
intel/libva#8  0x00007fffed679fa9 in GmmLib::GmmClientContext::DestroyResInfoObject (this=0x7a81f0, pResInfo=0x7fffd874e580) at /home/robinhe/media/gmmlib/Source/GmmLib/GlobalInfo/GmmClientContext.cpp:466
intel/libva#9  0x00007fffee93c3f2 in DdiMediaUtil_FreeSurface (surface=0x7fffd874e4c0) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva_util.cpp:998
intel/libva#10 0x00007fffee912e2c in DdiMedia_DestroySurfaces (ctx=0x791d70, surfaces=0x7fffe77fd1c4, num_surfaces=1) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva.cpp:2170
intel/libva#11 0x00007fffee91c8e0 in DdiMedia_GetImage (ctx=0x791d70, surface=21, x=0, y=0, width=64, height=64, image=0) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva.cpp:4884
intel/libva#12 0x00007ffff49814e3 in vaGetImage (dpy=0x791c00, surface=21, x=x@entry=0, y=y@entry=0, width=width@entry=64, height=height@entry=64, image=0) at ../va/va.c:1790
intel/libva#13 0x00007ffff5036af3 in gst_vaapi_surface_get_image (surface=0x7fffd8007b50, image=image@entry=0x7fffe844c5f0) at ../gst-libs/gst/vaapi/gstvaapisurface.c:714
intel/libva#14 0x00007ffff4ff7e92 in extract_allowed_surface_formats (direction=GST_PAD_SRC, specified_format=GST_VIDEO_FORMAT_NV12, img_formats=0x7fffe8006f00, display=0x72f100)
    at ../gst/vaapi/gstvaapipluginbase.c:1461
intel/libva#15 ensure_allowed_raw_caps (plugin=plugin@entry=0x777930, format=GST_VIDEO_FORMAT_NV12, direction=direction@entry=GST_PAD_SRC) at ../gst/vaapi/gstvaapipluginbase.c:1506
intel/libva#16 0x00007ffff4ff98ad in ensure_allowed_raw_caps (direction=GST_PAD_SRC, format=<optimized out>, plugin=0x777930, plugin@entry=0x0) at ../gst/vaapi/gstvaapipluginbase.c:1560
intel/libva#17 gst_vaapi_plugin_base_get_allowed_srcpad_raw_caps (plugin=plugin@entry=0x777930, format=<optimized out>) at ../gst/vaapi/gstvaapipluginbase.c:1557
intel/libva#18 0x00007ffff4ff4340 in gst_vaapidecode_ensure_allowed_srcpad_caps (decode=0x777930) at ../gst/vaapi/gstvaapidecode.c:228

``` 

It seems that when when the image and surface have different formats and then an inside VP pipeline is triggered, this will happen.  The surface 64x64 can duplicate this while 8x8 is OK. Bisect the commit ID is:
```
ebc16f89affb110bd8d15c1ad28ed824ad8f13b8 is the first bad commit
commit ebc16f89affb110bd8d15c1ad28ed824ad8f13b8
Author: Wang_PengX <peng.x.wang@intel.com>
Date:   Fri Feb 28 05:13:10 2020 +0800

    [VP] Use TileY on VP write to improve performance
    
    Use TileY on VP write to improve performance on linux
    
    Change-Id: Ief060deada57574ba3b1e26404e46ac4c8f5ac27

```

Seems to be a tiling alignment related problem.

HeJunyan, 2020-03-02:
sorry, transferred.

uartie, 2020-03-02:
@XinfengZhang can you please revert ebc16f89affb110bd8d15c1ad28ed824ad8f13b8 until someone can investigate the problem and a better solution?  This commit breaks most of our test cases.

XinfengZhang, 2020-03-03:
@FurongZhang , hi Furong, please consider to revert the commit and estimate the impact

XinfengZhang, 2020-03-04:
reverted by https://github.com/intel/media-driver/commit/9d1717e4be4a07c0f42df9419dc27f471b98756c

closed 2020-03-30 by FurongZhang