intel/media-driver #867: GPU Hang When FFmpeg-vappi APP-->transpose On EHL Platform

dryjq, 2020-03-05:
When ffmpeg vaapi deal with  video post-processing(VPP) on EHL platform, it is found that some cases fail due to GPU hang.

test case 1:
CMD：ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt nv12 -s:v 1920x1080 -i /home/labuser/focus/work_space/media_asset_nfs/yuv/1080p_blue_sky_1920x1080_NV12.yuv -vf 'format=nv12|vaapi,hwupload,transpose_vaapi=dir=clock,hwdownload,format=nv12' -pix_fmt nv12 -f rawvideo -vsync passthrough -an -vframes 10 -y /home/labuser/focus/work_space/build/NV12.yuv

ERROR: [graph 0 input from stream 0:0 @ 0x55566c133f40] w:1920 h:1080 pixfmt:nv12 tb:1/25 fr:25/1 sar:0/1
[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[AVHWFramesContext @ 0x55566c12ea40] Failed to sync surface 0xa: 34 (HW busy now).
[hwdownload @ 0x55566c132b00] Failed to download frame: -5.
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x55566c12e840] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x55566c1291c0] Statistics: 3110400 bytes read, 0 seeks
Conversion failed!
root@ssp-ehl-CDI376:/home/labuser/focus/work_space/build# dmesg
[5367473.451546] i915 0000:00:02.0: Resetting vecs0 for preemption time out
[5427411.503484] i915 0000:00:02.0: Resetting vecs0 for preemption time out

test case 2:
CMD：ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt nv12 -s:v 1920x1080 -i /home/labuser/focus/work_space/media_asset_nfs/yuv/1920x1080.nv12 -vf 'format=nv12|vaapi,hwupload,transpose_vaapi=dir=reversal,hwdownload,format=nv12' -pix_fmt nv12 -f rawvideo -vsync passthrough -an -vframes 1 -y /home/labuser/focus/work_space/build/NV12.yuv

ERROR：[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[AVHWFramesContext @ 0x562c81c7ec00] Failed to sync surface 0xa: 34 (HW busy now).
[hwdownload @ 0x562c81c7dbc0] Failed to download frame: -5.
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x562c81c799c0] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x562c81c741c0] Statistics: 3110400 bytes read, 0 seeks
Conversion failed!

FurongZhang, 2020-03-05:
@dryjq  I checked the log. It seems that FFMpeg set color primaries as VAProcColorStandardExplicit, but did not set correct color primaries, driver only has a default value BT601.

FurongZhang, 2020-03-30:
Please check the latest iHD driver, we have already fixed it. @dryjq

FurongZhang, 2020-03-30:
We have verified it from iHD driver side. Hence, I will close this issue. If any concern or open, you can re-open it if needed. Thanks a lot in advance!

closed 2020-03-30 by FurongZhang