intel/media-driver #868: GPU Hang When FFmpeg-vappi APP-->rotation  On EHL Platform

dryjq, 2020-03-05:
Test case 1:
CMD：ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt nv12 -s:v 1920x1080 -i /home/labuser/focus/work_space/media_asset_nfs/yuv/1080p_blue_sky_1920x1080_NV12.yuv -vf 'format=nv12|vaapi,hwupload,transpose_vaapi=dir=cclock,hwdownload,format=nv12' -pix_fmt nv12 -f rawvideo -vsync passthrough -an -vframes 10 -y /home/labuser/focus/work_space/build/NV12.yuv
ERROR：[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[AVHWFramesContext @ 0x55ce7231ea40] Failed to sync surface 0xa: 34 (HW busy now).
[hwdownload @ 0x55ce72322b00] Failed to download frame: -5.
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x55ce7231e840] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x55ce723191c0] Statistics: 3110400 bytes read, 0 seeks
Conversion failed!

Test case 2:

CMD:
 ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt nv12 -s:v 1920x1080 -i /home/labuser/focus/work_space/media_asset_nfs/yuv/1920x1080.nv12 -vf 'format=nv12|vaapi,hwupload,transpose_vaapi=dir=cclock,hwdownload,format=nv12' -pix_fmt nv12 -f rawvideo -vsync passthrough -an -vframes 1 -y /home/labuser/focus/work_space/build/NV12.yuv

ERROR:[graph 0 input from stream 0:0 @ 0x5588809df080] w:1920 h:1080 pixfmt:nv12 tb:1/25 fr:25/1 sar:0/1
[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[VP]:   CRITICAL - VpSetColorStandardExplictly:1779: unknown Color Standard for YUV format.
[AVHWFramesContext @ 0x5588809dec00] Failed to sync surface 0xa: 34 (HW busy now).
[hwdownload @ 0x5588809ddbc0] Failed to download frame: -5.
Error while filtering: Input/output error
Failed to inject frame into filter network: Input/output error
Error while processing the decoded data for stream #0:0
[AVIOContext @ 0x5588809d99c0] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x5588809d41c0] Statistics: 3110400 bytes read, 0 seeks
Conversion failed!

root@ssp-ehl-CDI376:/home/labuser/focus/work_space/build# dmesg
[5430718.511701] i915 0000:00:02.0: Resetting vecs0 for preemption time out

FurongZhang, 2020-03-05:
The issue is same as https://github.com/intel/media-driver/issues/867.

FurongZhang, 2020-03-30:
We have verified it from iHD driver side. Hence, I will close this issue. If any concern or open, you can re-open it if needed. Thanks a lot in advance!

FurongZhang, 2020-03-30:
@dryjq , please help to verify it from your side. Thanks a lot in advance!

closed 2020-03-30 by FurongZhang