intel/media-driver #870: Werror issue on fedora

kwizart, 2020-03-09:
Here a Warning turned into when building on fedora.

> 
> CMakeFiles/iHD_drv_video_OBJ.dir/linux/common/os/i915/xf86drm.c.o -c /builddir/build/BUILD/media-driver-intel-media-19.4.0/media_driver/linux/common/os/i915/xf86drm.c
> /builddir/build/BUILD/media-driver-intel-media-19.4.0/media_driver/linux/common/os/i915/xf86drm.c: In function 'int drmOpenByBusid(const char*, int)':
> /builddir/build/BUILD/media-driver-intel-media-19.4.0/media_driver/linux/common/os/i915/xf86drm.c:606:15: error: '%s' directive argument is null [-Werror=format-overflow=]
>   606 |         drmMsg("drmOpenByBusid: drmGetBusid reports %s\n", buf);
>       |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> cc1plus: all warnings being treated as errors

dvrogozh, 2020-03-09:
We did not see this one building with gcc-9.2.1. Are you using gcc-10 which is under development? (I think I saw that in your logs)

dvrogozh, 2020-03-20:
w/ self built gcc-10 (head of gcc tree) I see this:
```
/opt/intel/git/media-driver/media_driver/linux/ult/libdrm_mock/xf86drm_mock.c: In function 'int drmOpenByBusid(const char*, int)':
/opt/intel/git/media-driver/media_driver/linux/ult/libdrm_mock/xf86drm_mock.c:815:15: warning: '%s' directive argument is null [-Wformat-overflow=]
  815 |         drmMsg("drmOpenByBusid: drmGetBusid reports %s\n", buf);
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...
/opt/intel/git/media-driver/media_driver/linux/common/os/i915/xf86drm.c: In function 'int drmOpenByBusid(const char*, int)':
/opt/intel/git/media-driver/media_driver/linux/common/os/i915/xf86drm.c:606:15: error: '%s' directive argument is null [-Werror=format-overflow=]
  606 |         drmMsg("drmOpenByBusid: drmGetBusid reports %s\n", buf);
      |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

PR #885 (2020-03-25): build: fix Werror=format-overflow under gcc-10
Fixes: #870

With upcoming gcc-10 we start to see Werror=format-overflow reporting
that we attempt to printf %s null strings.

Change-Id: I7a62bc98bf7dd6bcd61d475d7650a06aa1a70932
commit da966a2a (2020-03-25): build: fix Werror=format-overflow under gcc-10

Fixes: #870

With upcoming gcc-10 we start to see Werror=format-overflow reporting
that we attempt to printf %s null strings.

Change-Id: I7a62bc98bf7dd6bcd61d475d7650a06aa1a70932
commit 712fa54b (2020-04-08): build: fix Werror=format-overflow under gcc-10

Fixes: #870

With upcoming gcc-10 we start to see Werror=format-overflow reporting
that we attempt to printf %s null strings.

Change-Id: I7a62bc98bf7dd6bcd61d475d7650a06aa1a70932
closed 2020-03-25 by intel-mediadev