intel/media-driver #873: Mismatch of "intra_insert_size" for Rolling Intra Refresh feature

djie1, 2020-03-12:
In VAAPI, VAEncMiscParameterRIR::intra_insert_size is defined as "Indicates the number of columns or rows in MB", see https://github.com/intel/libva/blob/master/va/va.h#L2137,

but in iHD driver, it sets to "have an overlap of one MB row or column of Intra macroblocks",  see https://github.com/intel/media-driver/blob/master/media_driver/agnostic/gen9_skl/codec/hal/codechal_encode_avc_g9_skl.cpp#L2458 for example on G9 SKL.

So the result is that in the encoded stream with RIR enabled, the width of MB row or column of intra  MBs is 1 more than the specified "intra_insert_size" as specified in VAAPI.

djie1, 2020-03-12:
In the source file, there is comment says "Pass the same IntraRefreshUnit to the kernel w/o the adjustment by -1, so as to have an overlap of one MB row or column of Intra macroblocks across one P frame to another P frame, as needed by the RollingI algo", explains the reason for the assignment.
I don't understand why the "RollingI algo" requires overlap of one MB row of column. Even it's the case, it should be set by VAAPI application to VAEncMiscParameterRIR::intra_insertion_location to make the overlap, but not changing application's input inside the driver.

Submitted PR#875 for proposal of the fix.

Xiaogangli-intel, 2022-04-27:
Hi @djie1, I checked this issue/source code again, and synced with some engineers who familiar with this history, actually, this is not issue, pass IntraRefreshUnit to HW directly is what HW expected.\nNot sure if you still have concern about it, if yes, please feel free let us know. Thank you for your contribution.

Xiaogangli-intel, 2022-05-07:
Hi @djie1, I close this issue because it's not an issue as I descripted above. If you still have any concern about it, let me know please.

closed 2022-05-07 by Xiaogangli-intel

djie1, 2022-05-07:
> Hi @djie1, I checked this issue/source code again, and synced with some engineers who familiar with this history, actually, this is not issue, pass IntraRefreshUnit to HW directly is what HW expected. Not sure if you still have concern about it, if yes, please feel free let us know. Thank you for your contribution.

your reply didn't solve the issue I described.