intel/media-driver #874: [issuse of vpping on x86 whiskey platform]scaling one frame of nv12 format to yuyv display format memory. If the width and height of the display is relatively small, the display memory splash screen will appear

PiperWei, 2020-03-12:
Zoom one frame of nv12 data to yuyv display memory. If the width and height of the display is relatively small, the display memory splash screen will appear

Who has encountered this problem？ and how to Solve？ thanks

PiperWei, 2020-03-12:
Who has encountered this problem？ and how to Solve？ thanks

joojler, 2020-03-17:
which version of driver are you using? Is your problem similar to attached picture? I'm put a  1080p NV12 frame to a display with YUYV and scaling down size such as 200x200. beside the 200x200 area, other places in the display buffer is full of garbege. 
[image]

Jister, 2020-03-18:
Dst rectangle area is correct, but background is garbage, seems colorfill is not done. App should set the background color parameter, i.e. output_background_color in VAProcPipelineParameterBuffer, otherwise, driver won’t fill the background. Could you check whether app set the output_background_color parameter? Thanks

FurongZhang, 2020-03-30:
@PiperWei  Any progress on this? Please set  output_background_color in VAProcPipelineParameterBuffer as @Jister mentioned in the above message.

FurongZhang, 2020-04-07:
Pending on @PiperWei feedback.

joojler, 2020-04-09:
this issue has been solved by providing the background color to do the color fill of the surface, otherwise random data will be there.

closed 2020-04-10 by FurongZhang