intel/media-driver #876: Memory corruption caused by incorrect typecasting in KernelDll_BuildKernel_CmFc()

ykku16github, 2020-03-12:
When KernelDll_BuildKernel_CmFc() calls cm_fc_combine_kernels() (see [1]), one of the inputs is &dwEstimatedKernelSize.  It is a uint32_t pointer but is typecasted to a size_t pointer.

On platforms where size_t is a 64-bit integer, subsequent write operations using this size_t pointer results in 4 bytes of memory corruption.

[1] https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/kdll/hal_kerneldll.c#L5045

FurongZhang, 2020-03-30:
I have commented the code changes #877. I think we had better not to use size_t for coding style(recommend to use uint32_t, uint64_t etc.).

FurongZhang, 2020-03-30:
There is code fix from @ykku16github , we are discussing the code changes.

FurongZhang, 2020-04-07:
The PR has been merged. @ykku16github I will close the issue. Please feel free to open it if you still see any issues.

closed 2020-04-07 by FurongZhang