intel/media-driver #878: vaInitialize failed with error code -1 (unknown libva error),exit

WhyDoWeWonder, 2020-03-16:
I'm running on Arch Linux with a core i7-9850H. I've set $LIBVA_DRIVER_NAME to iHD but vainfo returns `vaInitialize failed with error code -1 (unknown libva error),exit`

FurongZhang, 2020-03-30:
Can you provide more information about this? It is so rough that I can not have the useful information.

FurongZhang, 2020-03-30:
And the below wiki page is for your information: https://github.com/intel/media-driver/wiki/How-to-build-and-use-media-driver%3F.
Except this, please also pay attention to the version of libva and iHD driver(the version for both of them needs to be consistent).

FurongZhang, 2020-04-07:
@WhyDoWeWonder  Do you still see the issues?

FurongZhang, 2020-04-07:
@WhyDoWeWonder May I know if your issues are resolved or not?

FurongZhang, 2020-04-21:
It has been 2 weeks. I did not get any response from the bug submitter. Hence, I will close the issue. Please feel free to reopen it if you still see the issue, thanks a lot in advance!

closed 2020-04-21 by FurongZhang