intel/media-driver #879: Fails to build with Clang 10 due to -Werror

jbeich, 2020-03-17:
```c++
$ CC=clang10 CXX=clang++10 cmake -GNinja /path/to/media-driver
$ ninja
[...]
media_driver/agnostic/common/os/mos_utilities.c:73:21: error: loop variable 'data' of type 'const std::__1::pair<const std::__1::basic_string<char>, std::__1::vector<PerfUtility::Tick, std::__1::allocator<PerfUtility::Tick> > *>' creates a copy from type 'const std::__1::pair<const std::__1::basic_string<char>, std::__1::vector<PerfUtility::Tick, std::__1::allocator<PerfUtility::Tick> > *>' [-Werror,-Wrange-loop-construct]
    for (const auto data : records)
                    ^
media_driver/agnostic/common/os/mos_utilities.c:73:10: note: use reference type 'const std::__1::pair<const std::__1::basic_string<char>, std::__1::vector<PerfUtility::Tick, std::__1::allocator<PerfUtility::Tick> > *> &' to prevent copying
    for (const auto data : records)
         ^~~~~~~~~~~~~~~~~
                    &
media_driver/agnostic/common/renderhal/renderhal.cpp:5221:73: error: expression does not compute the number of elements in this array; element type is 'short', not 'uint64_t' (aka 'unsigned long') [-Werror,-Wsizeof-array-div]
    dwCount         = sizeof(pKernelEntry->pCscParams->Matrix[0].Coeff) / sizeof(uint64_t);
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^
media_driver/agnostic/common/renderhal/renderhal.cpp:5221:73: note: place parentheses around the 'sizeof(uint64_t)' expression to silence this warning
```

FurongZhang, 2020-03-18:
Install Clang
$sudo apt-get install clang-8

configure Clang compiling
$ cmake ../media-driver -DMEDIA_VERSION="2.0.0" -DCMAKE_C_COMPILER=clang-8 -DCMAKE_CXX_COMPILER=clang++-8 -DBUILD_TYPE=release-internal

Building code
$make -j8 VERBOSE=TRUE           #VERBOSE to check the build log.

FurongZhang, 2020-03-18:
Please let us know if it works for you or not. I tried on my side, it is working.

jbeich, 2020-03-18:
I don't disagree Clang < 10 builds fine. What doesn't is Clang >= 10.

dvrogozh, 2020-03-20:
Building with self-compiled ~clang-10 (very tip of llvm master) there is a bunch of errors including the above one (built w/ `make -j1 -i`):
```
[  2%] Built target igfxcmrt
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_decoder.cpp.o
In file included from /opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:51:
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:19: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
    typedef struct
                  ^
                   PIC_MHW_PARAMS
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:13: note: type is not C-compatible due to this member declaration
    typedef struct
            ^~~~~~
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:84:7: note: type is given name 'PIC_MHW_PARAMS' for linkage purposes by this typedef declaration
    } PIC_MHW_PARAMS;
      ^
1 error generated.
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_decode_avc.cpp.o
In file included from /opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.cpp:28:
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:19: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
    typedef struct
                  ^
                   PIC_MHW_PARAMS
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:13: note: type is not C-compatible due to this member declaration
    typedef struct
            ^~~~~~
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:84:7: note: type is given name 'PIC_MHW_PARAMS' for linkage purposes by this typedef declaration
    } PIC_MHW_PARAMS;
      ^
1 error generated.
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_mmc_decode_avc.cpp.o
In file included from /opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_mmc_decode_avc.cpp:27:
In file included from /opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_mmc_decode_avc.h:30:
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:19: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
    typedef struct
                  ^
                   PIC_MHW_PARAMS
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:13: note: type is not C-compatible due to this member declaration
    typedef struct
            ^~~~~~
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:84:7: note: type is given name 'PIC_MHW_PARAMS' for linkage purposes by this typedef declaration
    } PIC_MHW_PARAMS;
      ^
1 error generated.
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/os/mos_utilities.c.o
/opt/intel/git/media-driver/media_driver/agnostic/common/os/mos_utilities.c:73:21: error: loop variable 'data' creates a copy from type 'const std::pair<const std::__cxx11::basic_string<char>, std::vector<PerfUtility::Tick, std::allocator<PerfUtility::Tick> > *>' [-Werror,-Wrange-loop-construct]
    for (const auto data : records)
                    ^
/opt/intel/git/media-driver/media_driver/agnostic/common/os/mos_utilities.c:73:10: note: use reference type 'const std::pair<const std::__cxx11::basic_string<char>, std::vector<PerfUtility::Tick, std::allocator<PerfUtility::Tick> > *> &' to prevent copying
    for (const auto data : records)
         ^~~~~~~~~~~~~~~~~
                    &
1 error generated.
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/renderhal/renderhal.cpp.o
/opt/intel/git/media-driver/media_driver/agnostic/common/renderhal/renderhal.cpp:5221:73: error: expression does not compute the number of elements in this array; element type is 'short', not 'uint64_t' (aka 'unsigned long') [-Werror,-Wsizeof-array-div]
    dwCount         = sizeof(pKernelEntry->pCscParams->Matrix[0].Coeff) / sizeof(uint64_t);
                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ^
/opt/intel/git/media-driver/media_driver/agnostic/common/renderhal/renderhal.cpp:5221:73: note: place parentheses around the 'sizeof(uint64_t)' expression to silence this warning
1 error generated.
[  2%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/gen8_bdw/media_interfaces/media_interfaces_g8_bdw.cpp.o
In file included from /opt/intel/git/media-driver/media_driver/agnostic/gen8_bdw/media_interfaces/media_interfaces_g8_bdw.cpp:27:
In file included from /opt/intel/git/media-driver/media_driver/agnostic/gen8_bdw/media_interfaces/media_interfaces_g8_bdw.h:49:
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:19: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
[... cut ...] or-linkage]
    typedef struct
                  ^
                   PIC_MHW_PARAMS
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:73:13: note: type is not C-compatible due to this member declaration
    typedef struct
            ^~~~~~
/opt/intel/git/media-driver/media_driver/agnostic/common/codec/hal/codechal_decode_avc.h:84:7: note: type is given name 'PIC_MHW_PARAMS' for linkage purposes by this typedef declaration
    } PIC_MHW_PARAMS;
      ^
```

jbeich, 2020-03-20:
> Building with self-compiled ~clang-10 (very tip of llvm master)

LLVM `master` branch switched to Clang 11 a few months ago, so it may have more warnings than Clang 10. However, I've only tested Clang 10 on FreeBSD.

dvrogozh, 2020-03-20:
> LLVM master branch switched to Clang 11 a few months ago, so it may have more warnings than Clang 10. However, I've only tested Clang 10 on FreeBSD.

Yep, but till I reproduce whatever you see on your side I am good to go to fix it.

PR #886 (2020-03-24): clang-10/11: fix compilation issues
Fixes: #879

This fixes:
 * -Wnon-c-typedef-for-linkage
 * -Wsizeof-array-div
 * -Wrange-loop-construct

Change-Id: I0d081070ce5abb4c707af767330141eb87748065
commit 4330a1a9 (2020-03-24): clang-10/11: fix compilation issues

Fixes: #879

This fixes:
 * -Wnon-c-typedef-for-linkage
 * -Wsizeof-array-div
 * -Wrange-loop-construct

Change-Id: I0d081070ce5abb4c707af767330141eb87748065
commit 6529865b (2020-04-08): clang-10/11: fix compilation issues

Fixes: #879

This fixes:
 * -Wnon-c-typedef-for-linkage
 * -Wsizeof-array-div
 * -Wrange-loop-construct

Change-Id: I0d081070ce5abb4c707af767330141eb87748065
closed 2020-03-24 by intel-mediadev