intel/media-driver #881: Cannot get libva working with vainfo on Debian-8  : "va_openDriver() returns 18" after "va_getDriverName() returns -1"

khennicb, 2020-03-18:
Hi,

I am trying to use the QuickSync driver MSDK on a debian-8, and so I need to build libva and some other things.
I am installing vainfo from the intel git repository to check if libva works, and this is what gives me errors.

Here is my configuration : 

I install the dependencies
```
coder490a:~$ apt install -y cmake openbox git python3 rsync pkg-config automake curl build-essential autoconf libtool libdrm-dev libX11-dev libgl1-mesa-dev libpciaccess-dev libx11-dev libgl1-mesa-glx  xutils-dev
```
Then I download the sources of libva and libva-utils

```
git clone https://github.com/intel/libva.git --branch 2.5.0
git clone https://github.com/intel/libva-utils.git --branch 2.5.0

```

And I build it
```
( mkcd libva && ./autogen.sh && make -j4 && make install )
export PKG_CONFIG_PATH=${PKG_CONFIG_PATH}:/usr/local/lib/pkgconfig
( mkcd libva-utils && ./autogen.sh && make -j4  && make install )
```
And finally I set the variables : 
```
export LIBVA_DRIVERS_PATH=/usr/local/lib/dri/
export LIBVA_DRIVER_NAME=iHD
```
Now, when i run ```vainfo```, here is what happen
```
coder490a:~$ vainfo
error: can't connect to X server!
libva info: VA-API version 1.5.0
libva info: va_getDriverName() returns -1
libva info: User requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_5
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
libva error: /usr/local/lib/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 18
vaInitialize failed with error code 18 (invalid parameter),exit
```

```
coder490a:~$ ldd `which vainfo`
	linux-vdso.so.1 (0x00007ffdeb55a000)
	libva-x11.so.2 => /usr/local/lib/libva-x11.so.2 (0x00007f428b28b000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f428af48000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f428ad36000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f428ab30000)
	libva-drm.so.2 => /usr/local/lib/libva-drm.so.2 (0x00007f428a92e000)
	libdrm.so.2 => /usr/lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f428a721000)
	libva.so.2 => /usr/local/lib/libva.so.2 (0x00007f428a4fb000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f428a4f6000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f428a336000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f428a114000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f428b4a8000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4289f10000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4289d09000)
```

Do you have any idea about what means having ```va_openDriver() returns 18``` after ```va_getDriverName() returns -1``` ?

I have added the current user (and root) to the group 'video' but it doesn't change anything.

zxye, 2020-03-18:
@khennicb which platform are you running on?

zxye, 2020-03-19:
@khennicb, if you have multiple graphics cards on your machine, make sure to use the correct device node:

$ dmesg|grep "\[drm\] Initialized"
[    3.800849] [drm] Initialized **i915** 1.6.0 20190619 for **0000:00:02.0** on minor 0
[    3.831126] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 1

$ ls -l /dev/dri/by-path/
total 0
lrwxrwxrwx 1 root root  8 Feb 25 03:30 pci-**0000:00:02.0**-card -> ../card0
lrwxrwxrwx 1 root root 13 Feb 25 03:30 pci-**0000:00:02.0**-render -> ../**renderD128**
lrwxrwxrwx 1 root root  8 Feb 25 03:30 pci-0000:01:00.0-card -> ../card1
lrwxrwxrwx 1 root root 13 Feb 25 03:30 pci-0000:01:00.0-render -> ../renderD129

$ vainfo --display drm --device /dev/dri/**renderD128**

zxye, 2020-03-20:
Close it as it's not a driver bug. Please let me know if you still see any issue after you select the correct dri device node. Thank you.

closed 2020-03-20 by zxye