intel/media-driver #887: Please explaind the purpose of the X.Y.Zr tags

sebastinas, 2020-03-21:
There seems to be a new scheme for naming tags. The latest release has been tagged as 20.1.0r instead of X.Y.Z as done earlier. Is this the new naming scheme for releases or what are those tags supposed to represent?

onabiull, 2020-03-23:
Generally, the scheme is the same. Background of r suffix: we had to include late-arrived patches into Q4 release. And tag was set too optimistically. This inevitably should have led to some confusion. So we had chosen an amended tag name over re-tagging. At the same time, we didn't want to use a hotfix naming scheme, i.e. 19.4.1. So 19.4.0r was born with r standing for release :). Btw, in 20.1 we expect a change above the current 20.1.0r tag.
Please follow the official release tags, marked as release on https://github.com/intel/media-driver/releases and sorry for the confusion.

XinfengZhang, 2020-03-24:
yes, misoperation, should no "r" in future, only X.Y.Z.

sebastinas, 2020-03-26:
My issue is that with this back and forth on tag names, I have to update the watch files we use to track releases in Debian and Ubuntu. For 19.4.0r we had to change them to include 19.4.0r, but now for 20.1.0r we're supposed to ignore them again. Note that our tooling sorts 20.1.0r after 20.1.0 (check dpkg --compare-versions), e.g. if I'd package 20.1.0r now I'd have to create some fake version for 20.1.0.

So, please stick to one naming scheme, and if there is a hotfix release don't use the same naming scheme for a prerelease. Thanks!

ArchangeGabriel, 2020-04-06:
Is the current 20.1.0 tag an actual release or not?

XinfengZhang, 2020-04-07:
@ArchangeGabriel , it is to correct previous one tag, and will release base on 20.1.1

closed 2020-04-09 by sebastinas