intel/media-driver #890: [regression] image corruption since commit 21e04f2ce5af496a3d9ff7ab0eb3f4f515cc1097

xhaihao, 2020-03-24:
I see corrupted image when rendering a RGBA surface since 21e04f2ce5af496a3d9ff7ab0eb3f4f515cc1097

$> GST_GL_PLATFORM=egl  gst-launch-1.0 videotestsrc ! vaapipostproc ! video/x-raw\(memory:DMABuf\),format=RGBA ! glimagesink

Before 21e04f2ce5af496a3d9ff7ab0eb3f4f515cc1097, the above command works well.

Jister, 2020-03-24:
This commit changed the tiling type for ARGB formats from tile linear to tile Y, I suppose the corruption is related with the tiling type change. Could you help to check what tiling type does mesa set?

xhaihao, 2020-03-30:
It should be tile Y as mesa calls GET_TILING to get tiling type. 

BTW  I see 21e04f2 was reverted in commit f9cd0af (Revert "[VP] Enable RGB Y tiling and RC Support in Render"), so closed this issue as fixed.

closed 2020-03-30 by xhaihao

Jister, 2020-03-30:
@xhaihao We'd better have further investigation on the issue, since later the tiling type change for RGB will be merged again, which may expose the issue again.