intel/media-driver #894: [TGL] GPU hang on hevc SCC decoder

HeJunyan, 2020-03-30:
GPU get hang when run MSDK cmd:
sample_decode h265 -i Bipred_8by8_A_Qualcomm_2.bit -o scc_out.yuv
It's a HEVC SCC main stream.

wangyan-intel, 2020-03-31:
@HeJunyan This is MSDK issue and has been fixed. Please try the latest MSDK.

HeJunyan, 2020-03-31:
It's a random hang, you need to run it several times. And once hang, never come back

XinfengZhang, 2020-05-27:
@wangyan-intel  , any update?

wangyan-intel, 2020-06-03:
@HeJunyan 
I checked it with the latest iHD driver. The issue is gone. Could you please double check it?
Thanks.

HeJunyan, 2020-06-03:
I checked and not hang now. If you can not find the root cause, we can reopen it when we find it again.

wangyan-intel, 2020-06-03:
I think some SCC decoding patches fix of iHD driver this issue recently.

closed 2020-06-03 by wangyan-intel