intel/media-driver #897: Invalid allocation for GST vaapi plugin in Gemini Lake

zcwang, 2020-04-07:
[GST-error-in-GLK_gary0406.log](https://github.com/intel/media-driver/files/4442608/GST-error-in-GLK_gary0406.log)

Use one of 2 GST commands to cause issue in GLK (J5005 CPU @ 1.50GHz).

- gst-launch-1.0 videotestsrc ! vaapih264enc ! filesink location=test.h264
- gst-launch-1.0 -ev videotestsrc num-buffers=600 ! video/x-raw,width=3840,height=2160,framerate=30/1 ! vaapih264enc ! h264parse ! mp4mux ! filesink location=test.mp4

...
libva info: VA-API version 1.7.0                                                                                                                                                                                                                                      [576/1882]
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /home/intel/test-media/vaapi/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.7 (libva 2.7.0.pre1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.2.pre (e96c2b6f)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
intel@intel-Geminilake:~/test-media$ gst-launch-1.0 videotestsrc ! vaapih264enc ! filesink location=test.h264
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
Got context from element 'vaapiencodeh264-0': gst.vaapi.Display=context, gst.vaapi.Display=(GstVaapiDisplay)"\(GstVaapiDisplayDRM\)\ vaapidisplaydrm1";
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.062344607 30555 0x55570123b1e0 ERROR            vaapiencode gstvaapiencode.c:196:gst_vaapiencode_default_alloc_buffer: invalid GstVaapiCodedBuffer size (0 bytes)
0:00:00.062383962 30555 0x55570123b1e0 ERROR            vaapiencode gstvaapiencode.c:303:gst_vaapiencode_push_frame: failed to allocate encoded buffer in system memory
0:00:00.066807014 30555 0x55570123b1e0 ERROR            vaapiencode gstvaapiencode.c:196:gst_vaapiencode_default_alloc_buffer: invalid GstVaapiCodedBuffer size (0 bytes)
0:00:00.066826192 30555 0x55570123b1e0 ERROR            vaapiencode gstvaapiencode.c:303:gst_vaapiencode_push_frame: failed to allocate encoded buffer in system memory
0:00:00.070649348 30555 0x55570123b1e0 ERROR            vaapiencode gstvaapiencode.c:196:gst_vaapiencode_default_alloc_buffer: invalid GstVaapiCodedBuffer size (0 bytes)
...
I list reference code commits of related code for my test environment,

- Media frameworks (red modules)
drm: " https://gitlab.freedesktop.org/mesa/drm.git"
0190f49a Bump the version to 2.4.100 for autotools

xf86-video-intel: “https://gitlab.freedesktop.org/xorg/driver/xf86-video-intel.git”
652d93cb i810: Avoid duplicate definition of I810PatternROP

gstreamer: "https://gitlab.freedesktop.org/gstreamer/gstreamer.git"
5c74c8c4f filesink: Fix crash caused by zero-size memory allocation

gstreamer-vaapi: "https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi.git"
a28046a8 libs: encoder: h265: support ICQ/QVBR BRC

gst-plugins-base: "https://gitlab.freedesktop.org/gstreamer/gst-plugins-base.git"
caca46e0e subparse: convert from pango-markup to utf8

gst-plugins-good: "https://gitlab.freedesktop.org/gstreamer/gst-plugins-good.git"
818b38ebd rtpjitterbuffer: fix waiting timer/queue code

gst-plugins-bsd: "https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad.git"
afb042cbb msdkenc: Try to find corresponding codec frame for encoded output buffer

FFmpeg: "https://github.com/FFmpeg/FFmpeg.git”
b5a8ad5623 ffmpeg: Print an error instead of a debug message on exit.

MediaSDK "https://github.com/Intel-Media-SDK/MediaSDK"
ce37d6c4 [h264e] Enable MaxSliceSize feature for Linux LowPower encoder

libva: "https://github.com/intel/libva.git"
ef8cbc9 add matrix_coefficients into VADecPictureParameterBufferAV1

libva-utils: "https://github.com/intel/libva-utils.git"
ab0806b test/createsurfaces: add some scoped traces

media driver: "https://github.com/intel/media-driver.git"
b080e65c Calculate the right offset when vaCreateImage need width or height alignment.

gmmlib: "https://github.com/intel/gmmlib.git"
c01f004 Add device ids new GFX devices

- i915 driver (green module), please download my built kernel for drm-tip with the latest gfx driver (3/30)

zcwang, 2020-04-08:
Add log for reproduced by GST-msdk (msdkh264enc) for reference. 

BTW knew GLK is non-POR in media-driver and MSDK by https://github.com/intel/media-driver#supported-platforms

...
intel@intel-Geminilake:~/test-media$ gst-launch-1.0 videotestsrc ! msdkh264enc ! filesink location=test.h264
Setting pipeline to PAUSED ...
libva info: VA-API version 1.7.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /home/intel/test-media/vaapi/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
Pipeline is PREROLLING ...
Got context from element 'msdkh264enc0': gst.msdk.Context=context, gst.msdk.Context=(GstMsdkContext)"\(GstMsdkContext\)\ msdkcontext0";
Redistribute latency...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstMsdkH264Enc:msdkh264enc0: Encode frame failed.
Additional debug info:
../sys/msdk/gstmsdkenc.c(1015): gst_msdkenc_encode_frame (): /GstPipeline:pipeline0/GstMsdkH264Enc:msdkh264enc0:
MSDK encode error (undefined error)
Execution ended after 0:00:00.001531363
Setting pipeline to NULL ...
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3075): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason error (-5)
ERROR: from element /GstPipeline:pipeline0/GstMsdkH264Enc:msdkh264enc0: Encode frame failed.
Additional debug info:
../sys/msdk/gstmsdkenc.c(1089): gst_msdkenc_flush_frames (): /GstPipeline:pipeline0/GstMsdkH264Enc:msdkh264enc0:
MSDK encode error (undefined error)
ERROR: from element /GstPipeline:pipeline0/GstMsdkH264Enc:msdkh264enc0: Encode frame failed.
Additional debug info:
../sys/msdk/gstmsdkenc.c(1089): gst_msdkenc_flush_frames (): /GstPipeline:pipeline0/GstMsdkH264Enc:msdkh264enc0:
MSDK encode error (undefined error)
0:00:00.187646136 20812 0x56245310e000 ERROR                msdkenc gstmsdkenc.c:926:gst_msdkenc_finish_frame:<msdkh264enc0> failed to get list of frame
0:00:00.187693302 20812 0x56245310e000 ERROR                msdkenc gstmsdkenc.c:926:gst_msdkenc_finish_frame:<msdkh264enc0> failed to get list of frame
Freeing pipeline ...
intel@intel-Geminilake:~/test-media$
...

XinfengZhang, 2020-10-19:
@zcwang hi Gary, have you tried PR #1005 ?

zcwang, 2020-10-21:
I used the latest code in upstream and  issued went away.  Closed issue.
Thanks!
Gary

closed 2020-10-21 by zcwang