intel/media-driver #898: [FFmpeg][HEVC][ENCODE][multi-slice]encode failed for hevc multislice

zhangyuankun-star, 2020-04-09:
Driver does not support encoding pictures as multiple slices.
os: ubuntu 18.0.4
platform:TGL
kernel : 5.6.0-rc6+
firmware:  https://kernel.googlesource.com/pub/scm/linux/kernel/git/firmware/linux-firmware.git
libva : 0e391fd490050f0b7492f5af38b86ba6cccfadc4
gmmlib: 09324e1fe8129b66bdf6b16ed533d56ce654eaa4
media-driver: 7a6308cd879b65550fe4de2c2086d3b9113d5b40
ffmpeg-cartwheel: https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/vaapi_encode.c#L1865 
CMD:
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt p010le -s:v 1920x1080 -r:v 30 -i /home/yuankun/media_assets_sanity_local/yuv/1080P_1920x1080.P010 -vf 'format=p010le,hwupload' -c:v hevc_vaapi -low_delay_b 1 -profile:v main10 -rc_mode CBR -g 30 -slices 2 -bf 2 -b:v 5000k -maxrate 5000k -low_power 0 -vframes 150 -y 1.h265
LOG：
ffmpeg version N-97288-g1794e38d40 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: --enable-vaapi --prefix=/opt/X11R7/iHD/ffmpeg_mainline --enable-nonfree --enable-shared --enable-libmfx --enable-gpl --enable-libx264 --enable-libx265 --enable-debug --extra-cflags=-g --extra-ldflags=-g
  libavutil      56. 42.102 / 56. 42.102
  libavcodec     58. 77.101 / 58. 77.101
  libavformat    58. 42.100 / 58. 42.100
  libavdevice    58.  9.103 / 58.  9.103
  libavfilter     7. 77.101 /  7. 77.101
  libswscale      5.  6.101 /  5.  6.101
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
[AVHWDeviceContext @ 0x55c973e638c0] libva: VA-API version 1.8.0
[AVHWDeviceContext @ 0x55c973e638c0] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x55c973e638c0] libva: Trying to open /opt/X11R7/iHD/vpg_driver/lib/dri//iHD_drv_video.so
[AVHWDeviceContext @ 0x55c973e638c0] libva: Found init function __vaDriverInit_1_8
[AVHWDeviceContext @ 0x55c973e638c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x55c973e638c0] Initialised VAAPI connection: version 1.8
[AVHWDeviceContext @ 0x55c973e638c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 20.2.pre (7a6308cd).
[AVHWDeviceContext @ 0x55c973e638c0] Driver not found in known nonstandard list, using standard behaviour.
[rawvideo @ 0x55c973ede500] Estimating duration from bitrate, this may be inaccurate
Input #0, rawvideo, from '/home/yuankun/media_assets_sanity_local/yuv/1080P_1920x1080.P010':
  Duration: 00:00:10.00, start: 0.000000, bitrate: 1492992 kb/s
    Stream #0:0: Video: rawvideo, 1 reference frame, p010le, 1920x1080, 1492992 kb/s, 30 tbr, 30 tbn, 30 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> hevc (hevc_vaapi))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x55c973eeff00] w:1920 h:1080 pixfmt:p010le tb:1/30 fr:30/1 sar:0/1
[hevc_vaapi @ 0x55c973eeb6c0] Input surface format is p010le.
[hevc_vaapi @ 0x55c973eeb6c0] Using VAAPI profile VAProfileHEVCMain10 (18).
[hevc_vaapi @ 0x55c973eeb6c0] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[hevc_vaapi @ 0x55c973eeb6c0] Using VAAPI render target format YUV420_10 (0x100).
[hevc_vaapi @ 0x55c973eeb6c0] RC mode: CBR.
[hevc_vaapi @ 0x55c973eeb6c0] RC target: 100% of 5000000 bps over 1000 ms.
[hevc_vaapi @ 0x55c973eeb6c0] RC buffer: 5000000 bits, initial fullness 3750000 bits.
[hevc_vaapi @ 0x55c973eeb6c0] RC framerate: 30/1 (30.00 fps).
[hevc_vaapi @ 0x55c973eeb6c0] Using intra, P- and B-frames (supported references: 4 / 4).
[hevc_vaapi @ 0x55c973eeb6c0] Driver does not support encoding pictures as multiple slices.
.Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
[AVIOContext @ 0x55c973eec640] Statistics: 0 seeks, 0 writeouts
[AVIOContext @ 0x55c973edefc0] Statistics: 6220800 bytes read, 0 seeks
Conversion failed!

fulinjie, 2020-04-09:
Root cause inside driver, missing max slice query support in gen 12:
https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L1437
Hi @XinfengZhang, would you please help to take a look?

XinfengZhang, 2020-06-03:
@fulinjie  , not very clear, are you talking about https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L1458?

fulinjie, 2020-06-03:
> @fulinjie , not very clear, are you talking about https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp#L1458?

Yep

PR #961 (2020-06-09): add HEVC VME MaxSlice caps
Fixes: #898
commit 152008b6 (2020-06-09): add HEVC VME MaxSlice caps

Fixes: #898
closed 2020-06-09 by intel-mediadev