intel/media-driver #900: mapped /dev/dri/renderD128 is never been unmapped

LindaYu17, 2020-04-10:
when testing with below python script(change .txt to .py) for video playback with vaapih264dec, we find that /dev/dri/renderD128 is mapped(drm_mmap), but even Gst.Pipeline.set_state(Gst.State.NULL) is call, /dev/dri/renderD128 is still mapped (can be seen with pmap -x -d PID | grep renderD128) until test program exit. If we stop and start media pipeline frequently without main process exit, its pmap will accumulate.  
what we expect is if Gst.Pipeline.set_state(Gst.State.NULL) is called, all the resource the pipeline owned is released.
[pmap-leak.txt](https://github.com/intel/media-driver/files/4459791/pmap-leak.txt)

And i add some log in driver and after some debug, I find that mapped gem_handle is conflict with previous vaCreateSurface 
(68 in the attached log snippet) and is destroyed soon and reused. 
[image]

Thanks for your support

FurongZhang, 2020-04-10:
I will check that ASAP.

LindaYu17, 2020-04-10:
by the way, I reproduce it on SKL and APL both and the driver version is https://github.com/Intel-Media-SDK/MediaSDK/releases/tag/intel-mediasdk-19.4.0

FurongZhang, 2020-04-14:
https://gitlab.freedesktop.org/mesa/drm/-/issues/40

FurongZhang, 2020-04-14:
This issue is originally in DRM. I have already filed an issue against DRM https://gitlab.freedesktop.org/mesa/drm/-/issues/40. Media driver just wrap the DRM functions.

closed 2020-04-28 by intel-mediadev