intel/media-driver #901: There is a problem with the cmake environment path setting

kppomzh, 2020-04-12:
After I run the command like "cmake ../media_driver/",it returns these errors:
[image]
These files do exist in the folder, but the path shown here is obviously incorrect.
In addition,when I built version 19.4.0,I could only using "cmake .."to build and install.It worked very well.

XinfengZhang, 2020-04-12:
hi ,  cmake ../media_driver/?? the CMake top file should located in media-driver/

kppomzh, 2020-04-12:
@XinfengZhang Well,I just follow the Readme.
[image]

closed 2020-07-16 by XinfengZhang