intel/media-driver #902: [EHL][FFmpeg][Enc] jpeg output size is 0.

feiwan1, 2020-04-13:
By using the cmd of ffmpeg-vaapi, the output jpeg file is 0:

$ ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug -f rawvideo -pix_fmt yuv420p -s:v 720x480 -i smpte_I420_720x480_I420.yuv -vf 'format=nv12,hwupload' -c:v mjpeg_vaapi -global_quality 100 -vframes 1 -y smpte_I420-cqp-10.jpg

This may caused by the check:
https://github.com/intel/media-driver/blob/57019bc881a5722defbddc74fdaf8786be0adfe1/media_driver/agnostic/common/codec/hal/codechal_encoder_base.cpp#L3826

There will be the error message of "missing huffman code", but the huffeman table is set through 
 vaBuffer. When I remove this check, the output file size will not be 0, and can be played well. Same cmd works on SKL/KBL/ICL.

media-driver: 57019bc881a5722defbddc74fdaf8786be0adfe1
ffmpeg: 932493f9c2a8cd9c79878694a26a0521114b59a4

feiwan1, 2020-04-13:
By the way, even here is an error, the returned status is SUCCESS from vaMapBuffer. So for application, we can't see any abnormal, until encode finish and then the output file size is 0. If can pass the error here to top va api, then we can detect it at once, no need to execute next process. Here is bt info:

DdiEncodeBase::StatusReport (this=0x6c2260, mediaBuf=0x732cb0, buf=0x7fffffffc1c8) at /root/build/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:163
(gdb) bt
#0  DdiEncodeBase::StatusReport (this=0x6c2260, mediaBuf=0x732cb0, buf=0x7fffffffc1c8) at /root/build/media-driver/media_driver/linux/common/codec/ddi/media_ddi_encode_base.cpp:163
#1  0x00007fffefa9d9e1 in DdiEncode_StatusReport (encCtx=0x6c8d90, mediaBuf=0x732cb0, buf=0x7fffffffc1c8) at /root/build/media-driver/media_driver/linux/common/codec/ddi/media_libva_encoder.cpp:165
#2  0x00007fffefb0bf28 in DdiMedia_MapBufferInternal (ctx=0x646cd0, buf_id=0, pbuf=0x7fffffffc1c8, flag=3) at /root/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3036
#3  0x00007fffefb0c573 in DdiMedia_MapBuffer (ctx=0x646cd0, buf_id=0, pbuf=0x7fffffffc1c8) at /root/build/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3188
#4  0x00007ffff38d7b61 in vaMapBuffer (dpy=0x646b60, buf_id=0, pbuf=0x7fffffffc1c8) at va.c:1438
#5  0x00007ffff5dd5439 in vaapi_encode_output (avctx=0x6a8440, pic=0x6c65c0, pkt=0x7fffffffc330) at libavcodec/vaapi_encode.c:591

Alexandr-Konovalov, 2020-05-26:
@feiwan1 , I unable to reproduce the issue on EHL, with ffmpeg/Media driver versions that you mentioned. Could you please share with me your input stream and ffmpeg logs?

dryjq, 2020-05-27:
The code stream is in this link ：https://drive.google.com/open?id=1lhWd4X6jaCOT1JC58TNh0mj3nBYD0Qs9
Here is the ffmpeg log information:
 ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v debug -f rawvideo -pix_fmt yuv420p -s:v 720x480 -i /home/labuser/focus/work_space/media_asset_nfs/yuv/smpte_I420_720x480_I420.yuv -vf 'format=nv12,hwupload' -c:v mjpeg_vaapi -global_quality 100 -vframes 1 -y smpte_I420-cqp-10.jpg
ffmpeg version N-98325-g455b5fb383 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.2.1-9ubuntu2)
  configuration: --enable-vaapi --prefix=/opt/X11R7/iHD/ffmpeg_mainline --enable-nonfree --pkg-config-flags=--static --enable-libmfx --enable-gpl --enable-libx264 --enable-debug=3 --disable-optimizations --disable-stripping --enable-shared --disable-static
  libavutil      56. 49.100 / 56. 49.100
  libavcodec     58. 87.101 / 58. 87.101
  libavformat    58. 43.100 / 58. 43.100
  libavdevice    58.  9.103 / 58.  9.103
  libavfilter     7. 83.100 /  7. 83.100
  libswscale      5.  6.101 /  5.  6.101
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
Splitting the commandline.
Reading option '-hwaccel' ... matched as option 'hwaccel' (use HW accelerated decoding) with argument 'vaapi'.
Reading option '-vaapi_device' ... matched as option 'vaapi_device' (set VAAPI hardware device (DRM path or X11 display name)) with argument '/dev/dri/renderD128'.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-s:v' ... matched as option 's' (set frame size (WxH or abbreviation)) with argument '720x480'.
Reading option '-i' ... matched as input url with argument '/home/labuser/focus/work_space/media_asset_nfs/yuv/smpte_I420_720x480_I420.yuv'.
Reading option '-vf' ... matched as option 'vf' (set video filters) with argument 'format=nv12,hwupload'.
Reading option '-c:v' ... matched as option 'c' (codec name) with argument 'mjpeg_vaapi'.
Reading option '-global_quality' ... matched as AVOption 'global_quality' with argument '100'.
Reading option '-vframes' ... matched as option 'vframes' (set the number of video frames to output) with argument '1'.
Reading option '-y' ... matched as option 'y' (overwrite output files) with argument '1'.
Reading option 'smpte_I420-cqp-10.jpg' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option vaapi_device (set VAAPI hardware device (DRM path or X11 display name)) with argument /dev/dri/renderD128.
[AVHWDeviceContext @ 0x559c9c9355c0] libva: VA-API version 1.8.0
[AVHWDeviceContext @ 0x559c9c9355c0] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x559c9c9355c0] libva: Trying to open /opt/X11R7/iHD/vpg_driver/lib/dri//iHD_drv_video.so
[AVHWDeviceContext @ 0x559c9c9355c0] libva: Found init function __vaDriverInit_1_8
[AVHWDeviceContext @ 0x559c9c9355c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x559c9c9355c0] Initialised VAAPI connection: version 1.8
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x41524742 -> bgra.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x42475241 -> argb.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x41424752 -> rgba.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x52474241 -> abgr.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x58524742 -> bgr0.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x42475258 -> 0rgb.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x58424752 -> rgb0.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x52474258 -> 0bgr.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30335241 -> unknown.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30334241 -> unknown.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30335258 -> x2rgb10le.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30334258 -> unknown.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x36314752 -> unknown.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x56555941 -> 0yuv.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30303859 -> gray.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x3132564e -> unknown.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x32595559 -> yuyv422.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x59565955 -> uyvy422.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x32315659 -> yuv420p.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30323449 -> yuv420p.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x50313134 -> yuv411p.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x48323234 -> yuv422p.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x56323234 -> yuv440p.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x50343434 -> yuv444p.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x33434d49 -> unknown.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30313050 -> p010le.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30313259 -> y210le.
[AVHWDeviceContext @ 0x559c9c9355c0] Format 0x30313459 -> y410le.
[AVHWDeviceContext @ 0x559c9c9355c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 20.2.pre (b69d534c).
[AVHWDeviceContext @ 0x559c9c9355c0] Driver not found in known nonstandard list, using standard behaviour.
Applying option v (set logging level) with argument debug.
Applying option y (overwrite output files) with argument 1.
[... cut ...]
  Output stream #0:0 (video): 1 frames encoded; 1 packets muxed (0 bytes);
  Total: 1 packets (0 bytes) muxed
Output file is empty, nothing was encoded (check -ss / -t / -frames parameters if used)
1 frames successfully decoded, 0 decoding errors
[mjpeg_vaapi @ 0x559c9c986d00] Freed output buffer 0
[AVIOContext @ 0x559c9c9802c0] Statistics: 518400 bytes read, 0 seeks

Last here is the commit information:
drive commit : b69d534c7ca90c50ce87216797b1e3a3b192949d
MSDK commit 984d9999ef9f6cb121784fc0c3f48b9f138d390b
FFmpeg commit  d5f87f85675cd7893200590a7d123622878fd9fc

feiwan1, 2020-05-27:
@dryjq  thanks for your update. @Alexandr-Konovalov this is EHL specific issue, other platforms all looks good. And for stream, we tested several different streams all can reproduce this issue.

feiwan1, 2020-09-18:
Verified by @xinyudox with #989, the output jpeg looks good. Close this issue. Thanks.

closed 2020-09-18 by feiwan1