intel/media-driver #903: [VP][regression] ~75-80% performance decrease in overlay/composite

uartie, 2020-04-14:
I am seeing a ~75-80% decrease in FPS performance with VP overlay/composite filter with recent iHD driver on my KBL.  It seems to have started with commit 922195ddd0ac.

My Current Software Stack:
```
libva (master) heads/master-0-ge18a6ab6861f 
gmmlib (master) tags/intel-gmmlib-20.1.1-0-g09324e1fe812 
intel-media-driver (master) heads/master-0-g3cb09c27e1af 
msdk (master) heads/master-0-g9fe77989675e 
gstreamer (master) heads/master-0-g88f0312624fe 
gst-plugins-base (master) heads/master-0-g7e5368627e4c 
gst-plugins-good (master) heads/master-0-gfd7ecac7935c 
gst-plugins-bad (master) heads/master-0-g462a8130a6cc 
gstreamer-vaapi (master) heads/master-0-gf1d13d1e0814 
```

Command:
```
gst-launch-1.0 -vf videotestsrc ! video/x-raw,format=NV12,width=3840,height=2160 \
 ! vaapipostproc saturation=1.1 contrast=1.1 hue=2 \
 ! vaapioverlay \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1" sync=false text-overlay=false
```

Command Output (some logging snipped), we get **~11 FPS**:
```
last-message = rendered: 7, dropped: 0, current: 12.98, average: 12.98
last-message = rendered: 13, dropped: 0, current: 11.44, average: 12.22
last-message = rendered: 19, dropped: 0, current: 11.50, average: 11.98
last-message = rendered: 25, dropped: 0, current: 11.31, average: 11.81
last-message = rendered: 31, dropped: 0, current: 11.40, average: 11.73
last-message = rendered: 37, dropped: 0, current: 10.98, average: 11.60
last-message = rendered: 43, dropped: 0, current: 11.21, average: 11.55
last-message = rendered: 49, dropped: 0, current: 10.77, average: 11.44
last-message = rendered: 55, dropped: 0, current: 11.22, average: 11.42
last-message = rendered: 61, dropped: 0, current: 11.28, average: 11.40
```

If I revert the following commits (top commits must revert for clean revert of 922195ddd0ac)...:
```
03fa8e62d4be - 2020-04-13 19:03:18 +0800 - YanfengMi       - [VP] Fix EHL/JSL alpha change issue 
171e3cc8dc0b - 2020-04-10 10:52:02 +0800 - ChenJasonK      - [VP] Remove condition for external surface. 
c9399690991b - 2020-03-31 21:58:50 +0800 - ChenJasonK      - [VP] Avoid cp external surface cases failure 
922195ddd0ac - 2020-03-25 15:23:02 +0800 - Chen_JasonK     - [VP] Add external surface support
```

Command Output with above commits reverted (some logging snipped), we get **~52 FPS**:
```
last-message = rendered: 25, dropped: 0, current: 48.15, average: 48.15
last-message = rendered: 52, dropped: 0, current: 53.65, average: 50.85
last-message = rendered: 79, dropped: 0, current: 53.44, average: 51.71
last-message = rendered: 105, dropped: 0, current: 50.96, average: 51.52
last-message = rendered: 133, dropped: 0, current: 55.43, average: 52.30
last-message = rendered: 159, dropped: 0, current: 50.52, average: 52.00
last-message = rendered: 184, dropped: 0, current: 49.39, average: 51.63
```

uartie, 2020-04-14:
cc: @Jister

FurongZhang, 2020-04-14:
@Jister

Jister, 2020-04-14:
yes, the performance drop is caused by the copy method changing in vaGetImage, I will file a PR to fix it.

Jister, 2020-04-16:
@uartie please try #909

uartie, 2020-04-16:
@Jister yes, #909 fixes this.  Thanks!

uartie, 2020-04-17:
@Jister, #909 seems to fix this on KBL...

But, I'm seeing big performance issues with VP composite/blend on an Apollo Lake (Celeron J3455), too.  And #909 does not seem to fix it.  Others have reported similar issue on other APL variants (e.g. 
Atom E3950), https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/246

```
gst-launch-1.0 -vf videotestsrc ! video/x-raw,format=NV12,width=3840,height=2160 \
 ! vaapipostproc saturation=1.1 contrast=1.1 hue=2 \
 ! vaapioverlay \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1" sync=false text-overlay=false

last-message = rendered: 3, dropped: 0, current: 5.34, average: 5.34
last-message = rendered: 5, dropped: 0, current: 3.61, average: 4.48
last-message = rendered: 7, dropped: 0, current: 3.61, average: 4.19
last-message = rendered: 9, dropped: 0, current: 3.61, average: 4.05
```

If we remove the vaapioverlay element (i.e. no VP composite/blend), it performs quite well:
```
gst-launch-1.0 -vf videotestsrc ! video/x-raw,format=NV12,width=3840,height=2160 \
 ! vaapipostproc saturation=1.1 contrast=1.1 hue=2 \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1" sync=false text-overlay=false

last-message = rendered: 16, dropped: 0, current: 30.71, average: 30.71
last-message = rendered: 31, dropped: 0, current: 29.36, average: 30.04
last-message = rendered: 46, dropped: 0, current: 29.68, average: 29.92
last-message = rendered: 61, dropped: 0, current: 29.52, average: 29.82
```

Can you take a look?

Jister, 2020-04-20:
@uartie This new issue should not  be related with this regression. 
I just tried to reproduce these two cmds, I found that without vaapioverlay element, gstreamer only call vpp once to do saturation/contrast/hue, but with vaapioverlay element, gstreamer will call vpp twice, fist  do do saturation/contrast/hue, second call set the blending parameters, but do nothing(input only one layer, the output is binary same with input). 
I'm just wondering why you need "vaapioverlay" here? If it's really needed, gstreamer can combine two vpp calls into one, I think that will benefit the performance.

jbolduc01, 2020-04-20:
@Jister How do you combine two vpp calls into one? I would like to try it on my Atom E3950 platform.

uartie, 2020-04-20:
@Jister I think the point was missed for APL issue... and it's my fault for not explaining better.  The problem is not multiple calls to vpp.  To demonstrate, we can split the procamp operations (and other vpp ops) into multiple vpp calls and will only see a minor drop in framerate, for example:
```
gst-launch-1.0 -vf videotestsrc ! video/x-raw,format=NV12,width=3840,height=2160 \
 ! vaapipostproc saturation=1.1 \
 ! vaapipostproc contrast=1.1 \
 ! vaapipostproc hue=2 \
 ! vaapipostproc sharpen=1 \
 ! vaapipostproc brightness=0.5 \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1" sync=false text-overlay=false

last-message = rendered: 15, dropped: 0, current: 28.85, average: 28.85
last-message = rendered: 29, dropped: 0, current: 26.92, average: 27.88
last-message = rendered: 43, dropped: 0, current: 27.47, average: 27.75
last-message = rendered: 57, dropped: 0, current: 27.41, average: 27.66
```

...but once we have vpp blend in the pipeline, the framerate falls off a cliff regardless of one or more inputs.  The fact that I only use one input to vpp blend in my example is irrelevant and was only intended to demonstrate the real problem.

I think you are right, however, that this issue is not related to the initially reported regression... since the problem exists before the bisected changes mentioned in the [initial bug description](https://github.com/intel/media-driver/issues/903#issue-599267668).  I will open a new issue for the APL vpp blend problem.

uartie, 2020-04-20:
#914 filed for APL VPP blend issue.

closed 2020-04-20 by uartie

uartie, 2020-04-20:
> @Jister How do you combine two vpp calls into one? I would like to try it on my Atom E3950 platform.

@jbolduc01 short of developing your own application, I believe [MediaSDK's](https://github.com/Intel-Media-SDK/MediaSDK) `sample_vpp` does this.  Although `sample_vpp` is probably not the solution you will need for the live-feed use-case you're looking for, it may help you with general debug/analysis.