intel/media-driver #905: [VP] Artifacts observed in output surface

ykku16github, 2020-04-14:
Chrome OS has a test case in which the VP module is used to do blit transfers.  Using the same VP context, 3 separate input surfaces with different resolutions are submitted for blit transfers to 3 separate output surfaces.  The following VP dump shows that the output of the first 2 blit transfers are fine, but the output of the 3rd blit transfer shows a top green bar:
[202004132107_vp_dump_3_bad.tar.gz](https://github.com/intel/media-driver/files/4473370/202004132107_vp_dump_3_bad.tar.gz)

ykku16github, 2020-04-14:
Debugging led to the following source code:
https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal_renderer.cpp#L2065

The maxSrcRect member is initialized to {0, 0, 0, 0} when VphalRenderer is first instantiated.  But it never gets reset to {0, 0, 0, 0} after that.  So, if there are multiple calls to vaEndPicture(), maxSrcRect will store the largest resolution from those calls.

As a workaround, if I reset maxSrcRect in https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal_renderer.cpp#L1069 or in https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal.cpp#L556, the problem goes away.

Here's the VP dump with my workaround:
[202004132123_vp_dump_4_good.tar.gz](https://github.com/intel/media-driver/files/4473506/202004132123_vp_dump_4_good.tar.gz)

ykku16github, 2020-04-14:
@FurongZhang , @wangyan-intel 
At this point, I need some feedback on the following:
1.  Is resetting maxSrcRect the right thing to do?
2.  If #1 is yes, then where is the proper place to do this reset?  I want to say it's https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal.cpp#L556.

FurongZhang, 2020-04-15:
@ykku16github  From your analysis, it seems that resetting maxSrcRect is the right thing to do. I agree with you to add the code in https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal.cpp#L556.
But I have a question, just now, I check all the places which use rcMaxSrc and maxSrcRect, I found driver program min(surface resolution, maxSrc). So I am curious why green corruption occurs.

FurongZhang, 2020-04-15:
Please feel free to file PR if you think your code changes are fixing the issue, thanks a lot in advance!

FurongZhang, 2020-04-15:
@ykku16github

ykku16github, 2020-04-16:
@FurongZhang 
To show how maxSrcRect is causing the problem, I added some debug prints to print out the variables used by the following 2 lines of source code:
1.  https://github.com/intel/media-driver/blob/master/media_driver/agnostic/gen9/hw/mhw_vebox_g9_X.cpp#L1954
2.  https://github.com/intel/media-driver/blob/master/media_driver/agnostic/gen9/hw/mhw_vebox_g9_X.cpp#L1957

For the 1st blit transfer of a 640x368 image (which is fine), I see:
Input surface:
*pdwSurfaceHeight = 368, pSurfaceParam->dwHeight = 384, pSurfaceParam->rcMaxSrc.bottom = 368, wHeightAlignUnit = 2
*pdwSurfaceWidth = 640, pSurfaceParam->dwWidth = 640, pSurfaceParam->rcMaxSrc.right = 640, wWidthAlignUnit = 2
Output surface:
*pdwSurfaceHeight = 368, pSurfaceParam->dwHeight = 368, pSurfaceParam->rcMaxSrc.bottom = 368, wHeightAlignUnit = 2
*pdwSurfaceWidth = 640, pSurfaceParam->dwWidth = 640, pSurfaceParam->rcMaxSrc.right = 640, wWidthAlignUnit = 2

For the 2nd blit transfer of a 1280x720 image (which is fine), I see:
Input surface:
*pdwSurfaceHeight = 720, pSurfaceParam->dwHeight = 736, pSurfaceParam->rcMaxSrc.bottom = 720, wHeightAlignUnit = 1
*pdwSurfaceWidth = 1280, pSurfaceParam->dwWidth = 1280, pSurfaceParam->rcMaxSrc.right = 1280, wWidthAlignUnit = 2
Output surface:
*pdwSurfaceHeight = 720, pSurfaceParam->dwHeight = 720, pSurfaceParam->rcMaxSrc.bottom = 720, wHeightAlignUnit = 2
*pdwSurfaceWidth = 1280, pSurfaceParam->dwWidth = 1280, pSurfaceParam->rcMaxSrc.right = 1280, wWidthAlignUnit = 2

For the 3rd blit transfer of a 640x368 image (which is bad), I see:
Input surface:
*pdwSurfaceHeight = 384, pSurfaceParam->dwHeight = 384, pSurfaceParam->rcMaxSrc.bottom = 720, wHeightAlignUnit = 2
*pdwSurfaceWidth = 640, pSurfaceParam->dwWidth = 640, pSurfaceParam->rcMaxSrc.right = 1280, wWidthAlignUnit = 2
Output surface:
*pdwSurfaceHeight = 368, pSurfaceParam->dwHeight = 368, pSurfaceParam->rcMaxSrc.bottom = 720, wHeightAlignUnit = 2
*pdwSurfaceWidth = 640, pSurfaceParam->dwWidth = 640, pSurfaceParam->rcMaxSrc.right = 1280, wWidthAlignUnit = 2

These debug prints show that the maxSrcRect for the 3rd blit transfer is 1280x720, which actually came from the 2nd blit transfer.  As a result, VeboxAdjustBoundary() incorrectly calculates the input surface resolution to be 640x384.

I think you can also argue that pSurfaceParam->dwHeight for the input surface is wrong for the 3rd blit transfer.  That parameter contains the "aligned" height.  If that parameter contains the "real" height, then VeboxAdjustBoundary() would correctly calculate the input surface resolution to be 640x368.  But I still think maxSrcRect being set to 1280x720 is incorrect.  What do you think?

FurongZhang, 2020-04-21:
Yes, I agree with you maxSrcRect being set to 1280x720 is incorrect.

FurongZhang, 2020-04-21:
@kku16github, are you planning to file a PR to fix this issue? If no, I can do that. May I know your opinion? Thanks a lot in advance!

FurongZhang, 2020-04-21:
@ykku16github

MicheleL-Intel, 2020-04-21:
@ykku16github, while your patch may have fixed the video artifacts by zeroing out the maxSrcRect coordinates and letting VphalRenderer to redefine the coordinates in accordance with the new resolution, it does not satisfy Google's recommendation to recreate the context and VA surface as the proper way to handle dynamic changes in the metadata.

The log attached to [PR 905|https://github.com/intel/media-driver/issues/905] shows the same context ID, which means it preserves the same context throughout, only the height and width are modified in accordance with the next frame while all else, including pixel formats, remain unchanged.

Preserving the same context has less performance impact, however recreating the context and VA surface ensures that any metadata changes in the next frame are accounted for in the decode pipeline.  Consider hypothetically these sequential JPEG frames: 640 x 368 4:2:0, 1280 x 720 4:2:2 and 640 x 480 4:4:4 which would require the metadata to be updated accordingly.

In other words, your patch undoubtedly fixes the green line artifact, however it does not transfer the less obvious potential changes to the remaining metadata in subsequent frames, which could result to noticeable structural dissimilarities.

ykku16github, 2020-04-21:
@MicheleL-Intel 
Since the context is created by the app, your comment is suggesting that the Chrome OS test case needs to change, instead of the media driver.  Is that correct?

@FurongZhang 
What do you think?  The last comment is suggesting that the same VP context shouldn't be re-used.    When surfaces with different resolutions (and different attributes) are submitted, the media driver expects the app to destroy and re-create the VP context.  Is that correct?

ykku16github, 2020-04-21:
> @kku16github, are you planning to file a PR to fix this issue? If no, I can do that. May I know your opinion? Thanks a lot in advance!

@FurongZhang , let's see how this latest discussion goes.  There might not be a need to change the media driver if we agree that the app is not supposed to re-use the VP context.

MicheleL-Intel, 2020-04-22:
@ykku16github , at this time the CROS unit test does not appear to be at issue.  Google confirmed the failure is not reproducible by eliminating VPP from the decode pipeline, therefore suspects the root cause involves the VPP (driver).

The unit test instantiates an object to prepare the test environment, including establishing shared memory and DMA-buf for storing VA surface data, blit from VPP.

I'm still reviewing the unit test code, and have not found where the application is creating the context (as you mentioned), although Google mentions it.  I thought the better place will be for the driver to create the initial and subsequent contexts as needed, not only to efficiently maintain the flow of execution and portability, but it is the driver that requires intimate knowledge of the instance.

Would you mind sharing the code snippet where you found the context being created by the application?

ykku16github, 2020-04-22:
The context is created by the chromium VAAPI wrapper.  The vaCreateContext call is here:
https://chromium.googlesource.com/chromium/src.git/+/refs/heads/master/media/gpu/vaapi/vaapi_wrapper.cc#1549

FurongZhang, 2020-04-23:
@ykku16github @MicheleL-Intel  I think there are 2 topics in this thread. The below is my understanding. 

For the case mentioned in the above: 640x368(#1 blit) -> 1280x720 (#2 blit) -> 640x368(#3 blit), to support this case, from the API design point view, no more context needed(we could support resolution change in one context). From the evidence @ykku16github provoided, there is trending to be a bug in media driver, I am reproducing it on my side with our test tool. 

For the Chromium case, I read the code of Chromium, VaapiWrapper::CreateContextAndSurfaces create surface and context(that means, surface and context are created in one function, if you want to change to a surface with different resolution, you need to call this function.). I am not sure if Chromium will call VaapiWrapper::CreateContext(const gfx::Size& size) on the top of VaapiWrapper layer.  I only see VaapiWrapper::CreateContext is called in VaapiWrapper::CreateContextAndSurfaces, VaapiWrapper::CreateContextAndScopedVASurface.

MicheleL-Intel, 2020-04-23:
@FurongZhang, by "context", are referring to the VPP context or VA context?  My understanding is that VPP creates its own context but only once, and this context is reused until teardown.  The VPP context assumes the values from the VA context as part of its initialization.

I'm also wondering why DdiMedia_DestroySurfaces is being called multiple times before the VA context is destroyed.  I thought the surface cannot be destroyed until the associated context is torn down.  It appears the return status is not being checked to confirm the surface is indeed destroyed.  Is the surface not being destroyed as expected?

ykku16github, 2020-04-23:
@FurongZhang , thanks for your feedback.  I've gone ahead and created a PR.  Let's see how this discussion ends, and we'll decide whether to proceed with this PR.

FurongZhang, 2020-04-23:
@MicheleL-Intel  VaapiWrapper::CreateContext is calling vaCreateContext, that is a VA context. Yes, if going into VP driver, VP driver will create the specific GPU context(e.g. VEBOX context, Render Engine Context). One VA context could have several VP GPU contexts which depends on Workload.
May I know where you see DdiMedia_DestroySurfaces is being called multiple times before the VA context is destroyed? I want to check that. Thanks a lot in advance!

FurongZhang, 2020-04-23:
@ykku16github  All right.

MicheleL-Intel, 2020-04-23:
@FurongZhang , thanks for clarifying.  Referring to attached 202004132123_vp_dump_4_good.tar.gz from Kang, the instance when DdiMedia_DestroySurfaces is being called, presumably prior to destruction of the associated context, can be found at line #172, and it happened just before the resolution changed from [640 x 368] to [1280 x 720].  There was a DdiMedia_DestroyContext a few lines afterwards.  

Will the function call at this instance (line 172 of the log) immediately return a failure, since the context has not yet been destroyed?  Sorry, I have not mapped the log to the call sequence yet, and will let you know the exact source line as soon as I find it.

andrescj-chromium, 2020-04-24:
Hi! Thanks for dealing with this! I wonder if I can be of assistance navigating the Chromium side of things.

For the case that @MicheleL-Intel is looking into, we have two VA contexts: one for JPEG decoding and another for VPP. We destroy and recreate the JPEG decoding VA context and surface when the resolution changes (I don't know why... It's just how it was when I started working on this :). This happens here [1].

On the other hand, the VA context used for VPP is never destroyed.

What I gather from the discussion here is that there is no decision yet on whether the application is expected to recreate the VPP context in this situation. If the application is responsible, then I think we need to clarify under what circumstances a context recreation should happen at the application level. A resolution change is an example, but are there other situations?

[1] https://cs.chromium.org/chromium/src/media/gpu/vaapi/vaapi_jpeg_decoder.cc?l=321-338&rcl=e6a0338191aab910f112d484a461596a9221df6e

ykku16github, 2020-04-25:
@andrescj-chromium , thanks for your input.  Based on this comment (https://github.com/intel/media-driver/issues/905#issuecomment-618184981), the same VP context can be re-used for blitting surfaces with different resolutions.  So, for this particular scenario, there's no need for the application to recreate the VP context.

@FurongZhang , please correct me if I'm wrong.

FurongZhang, 2020-04-26:
@ykku16github I agree your code change in PR https://github.com/ykku16github/media-driver/commit/6e577faf751f68b6c756454b954cd531edb9dc78. But my question is, have you see any code in chromium about setting VAProcPipelineParameterBuffer::surface_region/output_region. This variable indicate the input/output rectangle of destination surface. 

The height of surface could be different from the rectangle, but VAProcPipelineParameterBuffer::surface_region/output_region should be set correctly for VAAPI user. 

I searched the current chromium code, I did not find where VAProcPipelineParameterBuffer is set. 

Thanks a lot in advance!

FurongZhang, 2020-04-26:
@andrescj-chromium Yes, in most of cases, there is no need for VP context to recreate if only resolution change(surface size is different).

FurongZhang, 2020-04-26:
@ykku16github  May I know if Chromium test can run on Linux Ubuntu? Is it a light weight test tool? I'd like to run these tool to check the behavior of driver.

Thanks a lot in advance!

ykku16github, 2020-04-26:
> @ykku16github I agree your code change in PR [ykku16github@6e577fa](https://github.com/ykku16github/media-driver/commit/6e577faf751f68b6c756454b954cd531edb9dc78). But my question is, have you see any code in chromium about setting VAProcPipelineParameterBuffer::surface_region/output_region. This variable indicate the input/output rectangle.
> 
> The height of surface could be different from the rectangle, but VAProcPipelineParameterBuffer::surface_region/output_region should be set correctly for VAAPI user.
> 
> I searched the current chromium code, I did not find where VAProcPipelineParameterBuffer is set.
> 
> Thanks a lot in advance!

@FurongZhang , the following function contains what you are looking for:
https://chromium.googlesource.com/chromium/src.git/+/refs/heads/master/media/gpu/vaapi/vaapi_wrapper.cc#2073

ykku16github, 2020-04-26:
> @ykku16github May I know if Chromium test can run on Linux Ubuntu? Is it a light weight test tool? I'd like to run these tool to check the behavior of driver.
> 
> Thanks a lot in advance!

@FurongZhang , yes, it's possible to run on Ubuntu.  I found the following instructions:
https://chromium.googlesource.com/chromium/src/+/master/docs/linux/build_instructions.md

In the above link, search for "unit_tests".  You should see instructions on how to build and run tests. 
 I've never executed these instructions myself though.  Hopefully, they work for you.

@andrescj-chromium , am I pointing @FurongZhang in the right direction?

FurongZhang, 2020-04-27:
@ykku16github  Thank you so much for the information.

FurongZhang, 2020-04-27:
I am trying to merge your PR into master branch. @ykku16github

FurongZhang, 2020-04-27:
And if your PR is merged, please kindly close this issue, thanks a lot in advance! And please feel free to let me know if you have more questions.

andrescj-chromium, 2020-04-27:
@FurongZhang For the sake of posterity: in addition to @ykku16github's instructions for running unit tests on Linux, you should do the following to be able to run VA-API tests:

1) gn args out/Default
Add the following options:
target_os = "chromeos"
use_vaapi = true

2) Build the tests. For example, the one relevant for this bug:
ninja -C out/Default -j 1440 jpeg_decode_accelerator_unittest

3) Run it:
out/Default/jpeg_decode_accelerator_unittest --gtest_filter="All/MjpegDecodeAcceleratorTest.ResolutionChange/DMABUF" --save_to_file

Note that we hard code the render node path to /dev/dri/renderD128.

[1] https://cs.chromium.org/chromium/src/media/gpu/vaapi/vaapi_wrapper.cc?l=400&rcl=9cdcb106b0bcfee75cf3e94576fd48c055cc4bd6

closed 2020-04-28 by intel-mediadev