intel/media-driver #906: -DCMAKE_CXX_FLAGS='-std=c++11' not honored anymore

th0ma7, 2020-04-14:
Trying to update [spksrc](https://github.com/SynoCommunity/spksrc) to migrate from 19.4.0r to 20.1.1 driver version.  Cross-compiling cmake doesn't honor anymore the argument passing for `-std=c++11` which used to be the case with 19.4.0r resulting in failure to build:
```
Scanning dependencies of target igfxcmrt
make[5]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
make[5]: Entering directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
[  0%] Building CXX object cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_printf_host.cpp.o
x86_64-pc-linux-gnu-g++: error: unrecognized command line option ‘-std=c++17’
cmrtlib/linux/CMakeFiles/igfxcmrt.dir/build.make:62: recipe for target 'cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_printf_host.cpp.o' failed
make[5]: *** [cmrtlib/linux/CMakeFiles/igfxcmrt.dir/__/agnostic/share/cm_printf_host.cpp.o] Error 1
make[5]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
CMakeFiles/Makefile2:103: recipe for target 'cmrtlib/linux/CMakeFiles/igfxcmrt.dir/all' failed
make[4]: *** [cmrtlib/linux/CMakeFiles/igfxcmrt.dir/all] Error 2
make[4]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
Makefile:151: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
Makefile:80: recipe for target 'intel-media_custom_compile' failed
make[2]: *** [intel-media_custom_compile] Error 2
make[2]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-driver'
../../mk/spksrc.depend.mk:44: recipe for target 'depend_target' failed
make[1]: *** [depend_target] Error 2
make[1]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk'
../../mk/spksrc.cross-cc.mk:137: recipe for target 'arch-apollolake' failed
make: [arch-apollolake] Error 2 (ignored)
```

th0ma7, 2020-04-14:
Reverting changes to reflect 19.4.0r behaviour to`cmrtlib/linux/CMakeLists.txt` pointing to this commie https://github.com/intel/media-driver/commit/9f69551f1204981c8d88bbaee1f995f1cb377702#diff-31af95d452fabf1327a117c511426be2 such as the following fixes the issue:
```
--- ../media-driver-intel-media-20.1.1/cmrtlib/linux/CMakeLists.txt	2020-04-08 11:27:57.000000000 -0400
+++ cmrtlib/linux/CMakeLists.txt	2019-12-26 10:28:03.000000000 -0500
@@ -21,26 +21,21 @@
 cmake_minimum_required(VERSION 2.8)
 include (${CMAKE_CURRENT_LIST_DIR}/cmrt_utils.cmake)
 
-
 set(EXECUTABLE_OUTPUT_PATH ${MDF_OPTION__OUTPUT_DIR})
 set(LIBRARY_OUTPUT_PATH    ${MDF_OPTION__OUTPUT_DIR})
 
 # Set up compile options that will be used for the Linux build
-set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++17 -fPIC -fpermissive -fstack-protector-all -Werror")
-if(CMAKE_SIZEOF_VOID_P EQUAL 8)
-   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -msse4.1")
-endif()
-set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -fno-strict-aliasing -D_FORTIFY_SOURCE=2")
-set(CMAKE_CXX_FLAGS_DEBUG   "${CMAKE_CXX_FLAGS_DEBUG} -D_DEBUG -D__DEBUG -O0")
-set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -std=c++17 -fPIC -fpermissive -fstack-protector-all -Werror")
-if(CMAKE_SIZEOF_VOID_P EQUAL 8)
-   set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -msse4.1")
-endif()
-set(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -fno-strict-aliasing -D_FORTIFY_SOURCE=2")
-set(CMAKE_C_FLAGS_DEBUG   "${CMAKE_C_FLAGS_DEBUG} -D_DEBUG -D__DEBUG -O0")
+if(CMAKE_COMPILER_IS_GNUCC)
+    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -msse4.1 -std=c++11 -fPIC -fpermissive -fstack-protector-all -Werror")
+    set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -fno-strict-aliasing -D_FORTIFY_SOURCE=2")
+    set(CMAKE_CXX_FLAGS_DEBUG   "${CMAKE_CXX_FLAGS_DEBUG} -D_DEBUG -D__DEBUG -O0")
+    set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -msse4.1 -std=c++11 -fPIC -fpermissive -fstack-protector-all -Werror")
+    set(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE} -fno-strict-aliasing -D_FORTIFY_SOURCE=2")
+    set(CMAKE_C_FLAGS_DEBUG   "${CMAKE_C_FLAGS_DEBUG} -D_DEBUG -D__DEBUG -O0")
 
-set(GCC_SECURE_LINK_FLAGS "-z relro -z now")
-set(CMAKE_SKIP_RPATH ON)
+    set(GCC_SECURE_LINK_FLAGS "-z relro -z now")
+    set(CMAKE_SKIP_RPATH ON)
+endif(CMAKE_COMPILER_IS_GNUCC)
 
 #include paths
 include_directories(${CMAKE_CURRENT_LIST_DIR}/../proprietary/share/)
```

th0ma7, 2020-04-19:
This change ocured with commit https://github.com/intel/media-driver/commit/9f69551f1204981c8d88bbaee1f995f1cb377702

th0ma7, 2020-04-20:
Lastly, cross-building for multiple arches using the following from Synology:
```
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
```

dvrogozh, 2020-04-20:
Hi @th0ma7, it is not quite clear how you are configuring driver build. Could you, please, provide:
1. cmake configuration command line w/ any custom flags you've passed to configure it
2. make command line with any custom flags to adjust compiler settings.
3. Version of gcc you are using.

Note, please, that gcc starts to officially support c++11 with version gcc-6 only...

th0ma7, 2020-04-20:
Indeed using a 4.9.3 gcc compiler which does honor `-std=c++11` flag and was able to build properly previous version of intel-media-driver up to 19.4.x

Globally my flags are:
```
CMAKE_ARGS = -DBUILD_TYPE=release
CMAKE_ARGS += -DCMAKE_CROSSCOMPILING=TRUE
CMAKE_ARGS += -DCMAKE_SYSTEM_NAME=Linux
CMAKE_ARGS += -DCMAKE_CXX_FLAGS=-std=c++11
```
The current behavious seems to force `-std=c++17` flag which indeed isn't supported by gcc 4.9.3

dvrogozh, 2020-04-20:
-std=c++17 is indeed absolutely unexpected - this is a bug from my perspective and needs to be fixed. C++17 first appears in gcc-6 as experimental feature and becomes official in gcc-8. Media driver code is supposed to be built with >=gcc-6 and -std=c++11 and the above slipped into the code incorrectly.

gcc < 6 has experimental support of -std=c++11 which means that if someone will step into the compiler bug it won't actually be fixed into the compiler. Media driver code is known to step into one such problem for gcc-4.8 which got fixed in gcc-4.9. So you are good from this perspective using gcc-4.9, but I don't know whether you can step in any other problem - you have a risk here.

th0ma7, 2020-04-20:
Thnx for the answer!
Until this gets solved I can easily work things up with my reverting patch posted above.

th0ma7, 2020-07-05:
Error as now disappeared.  I believe this was fixed through https://github.com/intel/media-driver/commit/34815b3d3f210241da7541e7b1beabb01607f9cf
Built successfully TAG `intel-media-20.2.pre4`.
Closing the issue.

closed 2020-07-05 by th0ma7