intel/media-driver #907: error: deleting ‘void*’ is undefined [-Werror]

th0ma7, 2020-04-14:
Compiling 20.1.1 driver with help from patch https://github.com/intel/media-driver/issues/906#issuecomment-613432901 eventually leads into a `error: deleting ‘void*’ is undefined [-Werror]`
```
[ 63%] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/gen12/codec/hal/codechal_hw_g12_X.cpp.o
In file included from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_os.h:32:0,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/codec/hal/codechal.h:29,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/codec/hal/codechal_hw.h:29,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/gen12/codec/hal/codechal_hw_g12_X.h:29,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/gen12/codec/hal/codechal_hw_g12_X.cpp:29:
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_utilities.h: In instantiation of ‘void MOS_DeleteUtil(_Ty&) [with _Ty = void*]’:
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/media_driver_next/agnostic/common/shared/features/media_feature_const_settings.h:56:13:   required from here
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_utilities.h:311:9: error: deleting ‘void*’ is undefined [-Werror]
         delete(ptr);
         ^
In file included from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_utilities.h:172:0,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_os.h:32,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/codec/hal/codechal.h:29,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/codec/hal/codechal_hw.h:29,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/gen12/codec/hal/codechal_hw_g12_X.h:29,
                 from /home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/gen12/codec/hal/codechal_hw_g12_X.cpp:29:
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/media_driver_next/agnostic/common/os/mos_utilities_next.h: In instantiation of ‘static void MosUtilities::MosDeleteUtil(_Ty&) [with _Ty = void*]’:
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_utilities.h:302:45:   required from ‘void MOS_DeleteUtil(_Ty&) [with _Ty = void*]’
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/media_driver_next/agnostic/common/shared/features/media_feature_const_settings.h:56:13:   required from here
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/media_driver_next/agnostic/common/os/mos_utilities_next.h:132:13: error: deleting ‘void*’ is undefined [-Werror]
             delete(ptr);
             ^
cc1plus: all warnings being treated as errors
media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/build.make:10238: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/gen12/codec/hal/codechal_hw_g12_X.cpp.o' failed
make[5]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/gen12/codec/hal/codechal_hw_g12_X.cpp.o] Error 1
make[5]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
CMakeFiles/Makefile2:162: recipe for target 'media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all' failed
make[4]: *** [media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/all] Error 2
make[4]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
Makefile:151: recipe for target 'all' failed
make[3]: *** [all] Error 2
make[3]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/build'
Makefile:81: recipe for target 'intel-media_custom_compile' failed
make[2]: *** [intel-media_custom_compile] Error 2
make[2]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-driver'
../../mk/spksrc.depend.mk:44: recipe for target 'depend_target' failed
make[1]: *** [depend_target] Error 2
make[1]: Leaving directory '/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk'
../../mk/spksrc.cross-cc.mk:137: recipe for target 'arch-apollolake' failed
make: [arch-apollolake] Error 2 (ignored)
```

th0ma7, 2020-04-18:
I was able to track down the commit where this issue comes from: https://github.com/intel/media-driver/commit/7cd3f463078a63b3fb06a40c42da5adfb640e95b
As this is a rather large commit I really don't know in which part of the code the issue is...
Help much appreciated.

th0ma7, 2020-04-20:
Lastly, cross-building for multiple arches using the following from Synology:
```
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
```

th0ma7, 2020-04-24:
@fenhu briging this to your attention as you are the author of this commit https://github.com/intel/media-driver/commit/7cd3f463078a63b3fb06a40c42da5adfb640e95b leading to a failure to build.  Help on this would be greatly appreciated.

fenhu, 2020-04-27:
Hi th0ma7.

 I tried it on my side and couldn't reproduce it: gcc version 5.5.0 20171010 (Ubuntu 5.5.0-12ubuntu1~16.04)
The error which you provided show that the compiler cause this issue. Would you try to update your compiler? 

From the error logs, it seems the delete(*void prt) can't work on your side. while I think it is the common function which compiler should support it.  
=====================================================
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_utilities.h: In instantiation of ‘void MOS_DeleteUtil(_Ty&) [with _Ty = void*]’:
/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/media_driver_next/agnostic/common/shared/features/media_feature_const_settings.h:56:13:   required from here
// media_feature_const_settings.h:56:13
    virtual ~MediaFeatureConstSettings() 
    {
        if (m_featureSetting != nullptr)
        {
            MOS_Delete(m_featureSetting);
        }
    }

/home/spksrc/git-ffmpeg-evansport/spksrc/cross/intel-media-sdk/work-apollolake-6.1/media-driver-intel-media-20.1.1/media_driver/agnostic/common/os/mos_utilities.h:311:9: error: deleting ‘void*’ is undefined [-Werror]
         delete(ptr);
void MOS_DeleteUtil(_Ty& ptr)
#endif
{
   ............
    if (ptr != nullptr)
    {
        MOS_AtomicDecrement(&MosMemAllocCounter);
        MOS_MEMNINJA_FREE_MESSAGE(ptr, functionName, filename, line);
        delete(ptr);
        ptr = nullptr;
    }
}

th0ma7, 2020-04-27:
Thnx for the feedback @fenhu but updating the compiler isn't possible until Synology releases the next DSM version 7.x.  Until then GCC 4.9.x under a cross-compiling environment is the only version I can work with.

Also I noticed in the code that the "delete" function is also being called by various other genX prior to failing only on gen12.  Something different must be done elsewhere so it always worked until now?

Although it is really easy to reproduce doing the following on Ubuntu:
```
git clone https://github.com/th0ma7/spksrc.git
cd spksrc
```
Install the dependencies:
```
$ sudo dpkg --add-architecture i386 && sudo apt-get update
$ sudo apt install build-essential debootstrap python-pip automake autogen libgmp3-dev libltdl-dev libunistring-dev libffi-dev libcppunit-dev libncurses-dev imagemagick libssl-dev pkg-config zlib1g-dev gettext git curl subversion check intltool gperf flex bison xmlto php expect libgc-dev mercurial cython lzip cmake swig libc6-i386 libmount-dev libpcre3-dev libbz2-dev
$ sudo pip install -U setuptools pip wheel httpie
```
Setup the environment and move to the `intel-media-driver` tree:
```
make setup
cd cross/intel-media-driver
make arch-x64
```

fenhu, 2020-04-28:
@th0ma7 
My change is only open source code on Opensource for Gen12. There are some refactor work to make the codes simpler.  And you error cause by the void* point can't be deleted directly on old GCC. if you cannot upgrade the GCC. you can remove **MOS_Delete(m_featureSetting);** as temp fixing. Current now it can't be used on Opensource codes.  and I will discuss with the codes owner to decide the finally resolution.

// media_feature_const_settings.h:56:13
virtual ~MediaFeatureConstSettings()
{
**if (m_featureSetting != nullptr)
{
MOS_Delete(m_featureSetting);
}**
}

th0ma7, 2020-04-28:
Thnx for the feedback!
With your advice I was able to create the following which for now fixes the issue and allowed me to succesfully build the new driver version:
```
diff -uprN ../media-driver-intel-media-20.1.1-ORIG/media_driver/media_driver_next/agnostic/common/shared/features/media_feature_const_settings.h ./media_driver/media_driver_next/agnos
tic/common/shared/features/media_feature_const_settings.h
--- ../media-driver-intel-media-20.1.1-ORIG/media_driver/media_driver_next/agnostic/common/shared/features/media_feature_const_settings.h	2020-04-08 11:27:57.000000000 -0400
+++ ./media_driver/media_driver_next/agnostic/common/shared/features/media_feature_const_settings.h	2020-04-27 21:15:14.881241383 -0400
@@ -53,7 +53,8 @@ public:
     {
         if (m_featureSetting != nullptr)
         {
-            MOS_Delete(m_featureSetting);
+            // Issue https://github.com/intel/media-driver/issues/907
+	    //MOS_Delete(m_featureSetting);
         }
     }
 
@@ -89,4 +90,4 @@ protected:
     void *m_featureSetting = nullptr;
 };
 
-#endif  // !__MEDIA_FEATURE_CONST_SETTINGS_H__
\ No newline at end of file
+#endif  // !__MEDIA_FEATURE_CONST_SETTINGS_H__
```

fenhu, 2020-04-28:
Just synced with this file's owner, the engineer will move the MOS_Delete(m_featureSetting) to son class where malloc(m_featureSetting). And then the m_featureSetting will not be void* and have the detail struct.   I will notify you when the change is merged.

th0ma7, 2020-04-28:
Great!  Note that I can give a try at the upcoming patch before merging to confirm it solves the issue.

fenhu, 2020-05-06:
@th0ma7 
issue should be fixed by below change, pls sync latest code to try. 
https://github.com/intel/media-driver/commit/88769b23bc0f6ffd813f9aa946b82456d2bd1d77

th0ma7, 2020-05-06:
The patch does fix the issue.
Thnx a lot!

closed 2020-05-06 by th0ma7