intel/media-driver #910: [VP8][regression] decode error and crash

abhishek4kumar, 2020-04-17:
Patch register VP8 decode reference frame (https://github.com/intel/media-driver/pull/705/)
causes decoder error and crash in chromium os when playing vp8 video.
Reverting this solves the issue , patch commenst mention this is as work around for ffmpeg .

abhishek4kumar, 2020-04-24:
@XinfengZhang what will be impact if we revert this patch?

XinfengZhang, 2020-04-24:
it will impact ffmpeg-vaapi, ffmpeg -vaapi will create a context and decode 1 key frame, then create another context to decode interframes.  it is WA for such usage.

abhishek4kumar, 2020-04-28:
@XinfengZhang in chromeos vp8 playback crashes , seems double free issue 

As in chromeos only one context for keyframe and decode interframe  , it should not crash 

#11 0x00007808d85842bb in mos_gem_bo_free (bo=0x260b51ea26c0) at ../../libva-intel-media-driver-9999/media_driver/linux/common/os/i915/mos_bufmgr.c:1202
#12 0x00007808d8584e85 in mos_gem_bo_unreference_final (bo=0x260b51ea26c0, time=571) at ../../libva-intel-media-driver-9999/media_driver/linux/common/os/i915/mos_bufmgr.c:1389
#13 0x00007808d8584f01 in mos_gem_bo_unreference_locked_timed (bo=0x260b51ea26c0, time=571) at ../../libva-intel-media-driver-9999/media_driver/linux/common/os/i915/mos_bufmgr.c:1400
#14 0x00007808d8586487 in mos_gem_bo_clear_relocs (bo=0x260b51e3fea0, start=0) at ../../libva-intel-media-driver-9999/media_driver/linux/common/os/i915/mos_bufmgr.c:2417
#15 0x00007808d85b1717 in GpuContextSpecific::SubmitCommandBuffer (this=0x260b50df5a00, osInterface=0x260b50e01000, cmdBuffer=0x7808d09dbfa0, nullRendering=false)
    at ../../libva-intel-media-driver-9999/media_driver/linux/common/os/mos_gpucontext_specific.cpp:1137
#16 0x00007808d859e644 in Mos_Specific_SubmitCommandBuffer (pOsInterface=0x260b50e01000, pCmdBuffer=0x7808d09dbfa0, bNullRendering=0)
    at ../../libva-intel-media-driver-9999/media_driver/linux/common/os/mos_os_specific.c:3888
#17 0x00007808d816d4b3 in CodechalDecodeVp8G11::DecodePrimitiveLevel (this=0x260b50e4c400)
    at ../../libva-intel-media-driver-9999/media_driver/agnostic/gen11/codec/hal/codechal_decode_vp8_g11.cpp:322
#18 0x00007808d7b27704 in CodechalDecode::Execute (this=0x260b50e4c400, params=0x260b50dfe410)
    at ../../libva-intel-media-driver-9999/media_driver/agnostic/common/codec/hal/codechal_decoder.cpp:1284
#19 0x00007808d84e046f in DdiMediaDecode::EndPicture (this=0x260b51692600, ctx=0x260b4ff21400, context=268435456)
    at ../../libva-intel-media-driver-9999/media_driver/linux/common/codec/ddi/media_ddi_decode_base.cpp:810
#20 0x00007808d84e87df in DdiDecode_EndPicture (ctx=0x260b4ff21400, context=268435456) at ../../libva-intel-media-driver-9999/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:153
#21 0x00007808d856076f in DdiMedia_EndPicture (ctx=0x260b4ff21400, context=268435456) at ../../libva-intel-media-driver-9999/media_driver/linux/common/ddi/media_libva.cpp:3369
#22 0x00007808d886c5e8 in vaEndPicture (dpy=0x260b4ff31780, context=268435456) at ../../libva-9999/va/va.c:1584

abhishek4kumar, 2020-04-29:
@XinfengZhang   i tried debugging and with below patch it fixes crash in chrome os and also ffmpeg 4.2 works fine

diff --git a/media_driver/linux/common/codec/ddi/media_ddi_decode_vp8.cpp b/media_driver/linux/common/codec/ddi/media_ddi_decode_vp8.cpp
index c16f564e91ce..b391d4f54a42 100644
--- a/media_driver/linux/common/codec/ddi/media_ddi_decode_vp8.cpp
+++ b/media_driver/linux/common/codec/ddi/media_ddi_decode_vp8.cpp
@@ -167,21 +167,21 @@ VAStatus DdiDecodeVP8::ParsePicParams(
         {
             DdiMedia_MediaSurfaceToMosResource(lastRefSurface, &m_resNoneRegLastRefFrame);
             m_ddiDecodeCtx->DecodeParams.m_presNoneRegLastRefFrame = &m_resNoneRegLastRefFrame;
-            RegisterRTSurfaces(&m_ddiDecodeCtx->RTtbl, lastRefSurface);
+            //RegisterRTSurfaces(&m_ddiDecodeCtx->RTtbl, lastRefSurface);
         }
         goldenRefSurface = DdiMedia_GetSurfaceFromVASurfaceID(mediaCtx, picParam->golden_ref_frame);
         if(goldenRefSurface)
         {
             DdiMedia_MediaSurfaceToMosResource(lastRefSurface, &m_resNoneRegGoldenRefFrame);
             m_ddiDecodeCtx->DecodeParams.m_presNoneRegGoldenRefFrame = &m_resNoneRegGoldenRefFrame;
-            RegisterRTSurfaces(&m_ddiDecodeCtx->RTtbl, lastRefSurface);
+            //RegisterRTSurfaces(&m_ddiDecodeCtx->RTtbl, lastRefSurface);
         }
         altRefSurface    = DdiMedia_GetSurfaceFromVASurfaceID(mediaCtx, picParam->alt_ref_frame);
         if(altRefSurface)
         {
             DdiMedia_MediaSurfaceToMosResource(lastRefSurface, &m_resNoneRegAltRefFrame);
             m_ddiDecodeCtx->DecodeParams.m_presNoneRegAltRefFrame = &m_resNoneRegAltRefFrame;
-            RegisterRTSurfaces(&m_ddiDecodeCtx->RTtbl, lastRefSurface);
+           // RegisterRTSurfaces(&m_ddiDecodeCtx->RTtbl, lastRefSurface);
         }
     }

PR #924 (2020-05-14): [VP] fix crash in vp8 playback.
fixes vp8 decoding crash in chromeos,
ffmpeg 4.2 vp8 decoding works fine
in kbl/cml/jsl 
fixes #910
closed 2020-05-14 by intel-mediadev