intel/media-driver #912: TGL: image corruption when sharing a 64x64 NV12 surface between media-driver and the 3rd library.

xhaihao, 2020-04-20:
Issue can be reproduced by running a command below:

$> gst-launch-1.0 -v videotestsrc ! video/x-raw,format=NV12,width=64,height=64 ! msdkh265enc ! msdkh265dec ! 'video/x-raw(memory:DMABuf)' ! glimagesink

media-driver: 147804708619c45d2e201dff16911e011eb6995c
gstreamer: 8fb1fe46200b7aac2716e6a2413fbb9c7042e901
gst-plugins-base: ac6b8058278dfd02601cc5bcac75caf95be54f9d
gst-plugins-bad: 8f0ceba25143abe76b87f197525aac6a9f66485c
MediaSDK:  c92b652bb0f9bb8504c54076a3a1db9fed3fa50e

The same command line works well on ICL.

FurongZhang, 2020-04-21:
@xhaihao, which the third party is? Do you have a rough thought which component it is in? If in VP, I will check that.

xhaihao, 2020-04-22:
@FurongZhang the buffer is shared between media and 3D drivers.

Jister, 2020-05-29:
Hi, @xhaihao could you try with the latest Gmmlib(d7a0586104096f3e2241e2a773c6bc41a9e2c422) and check whether the issue is still exist or not?

xhaihao, 2020-06-02:
@Jister it works well with gmmlib commit d7a0586, thx

closed 2020-06-02 by xhaihao