intel/media-driver #913: TGL: GPU hang when encoding a 64x64 VP9 video

xhaihao, 2020-04-20:
This issue can be reproduced by running the commands below:
```
$> sample_encode vp9 -p010 -msb10 -i input.p010.yuv -w 64 -h 64 -o output.ivf
```
```
$> sample_encode vp9 -nv12 -i input.nv12.yuv -w 64 -h 64 -o output.ivf
```
media-driver: 1478047
gstreamer: 8fb1fe46200b7aac2716e6a2413fbb9c7042e901
gst-plugins-base: ac6b8058278dfd02601cc5bcac75caf95be54f9d
gst-plugins-bad: 8f0ceba25143abe76b87f197525aac6a9f66485c
MediaSDK: c92b652bb0f9bb8504c54076a3a1db9fed3fa50e

The same commands work well on ICL.

Xiaogangli-intel, 2020-04-23:
@xhaihao thank you, could you provide linux kernel version?

xhaihao, 2020-04-24:
I used https://cgit.freedesktop.org/drm-tip on TGL.

XinfengZhang, 2020-05-27:
minimum resolution of vp9 vdenc should be 128x96, will change the capability to report it.

commit 3571637e (2020-06-17): [Encode] VP9 VDEnc minimal pic resolution added to caps

VP9 encoder caps for min supported resolution
Fixes #913

Change-Id: Ia7e4056fbf9d2f37109b7e74575578d0cb036e26
commit 0fdeb564 (2020-06-17): [Encode] VP9 VDEnc minimal supported resolution check

VP9 encoder minimal supported resolution check
Fixes #913

Change-Id: Id121f9c3f60a156d886980fa4af0685f46c0016a
commit 26b7723d (2020-06-28): [Encode] VP9 VDEnc minimal supported resolution check

VP9 encoder minimal supported resolution check
Fixes #913

Change-Id: Id121f9c3f60a156d886980fa4af0685f46c0016a
closed 2020-06-17 by intel-mediadev