intel/media-driver #914: [VP][Blend] APL VPP blend is slow

uartie, 2020-04-20:
Low framerate with VPP **blend** on an Apollo Lake (Celeron J3455). Others have reported similar issue on other APL variants (e.g. Atom E3950), https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/246.

As seen by comparing the examples below, the framerate is hardly affected by other VPP operations, but falls off a cliff when using VPP blend operation.

This is a follow-up to the discussion in https://github.com/intel/media-driver/issues/903#issuecomment-616696875.  The extreme framerate drop does not exist on KBL (after fix for #903), but still exists on APL (with same fix).  Does APL use different execution path in driver than KBL for VPP blend?  Could the implementation for APL have similar cause as #903 (copy method in vaGetImage)?

1. Generate a test input file:
```
gst-launch-1.0 -vf videotestsrc num-buffers=500 \
 ! video/x-raw,format=NV12,width=3840,height=2160 \
 ! vaapih264enc ! h264parse \
 ! filesink location=test-input.h264
```

2. Decode with VPP blend (one input)
```
gst-launch-1.0 -vf filesrc location=test-input.h264 \
 ! h264parse ! vaapih264dec \
 ! vaapioverlay \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1 display=drm" \
   sync=false text-overlay=false

last-message = rendered: 5, dropped: 0, current: 9.28, average: 9.28
last-message = rendered: 9, dropped: 0, current: 7.09, average: 8.16
last-message = rendered: 13, dropped: 0, current: 6.98, average: 7.76
last-message = rendered: 17, dropped: 0, current: 6.92, average: 7.54
```

3. Decode with VPP blend (two inputs-- one with VPP scaling)
```
gst-launch-1.0 -vf filesrc location=test-input.h264 \
 ! h264parse ! vaapih264dec ! tee name=decoded ! queue \
 ! vaapioverlay name=blend \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1 display=drm" \
   sync=false text-overlay=false decoded. \
 ! queue ! vaapipostproc \
 ! "video/x-raw(memory:VASurface)",width=1920,height=1080 ! blend.

last-message = rendered: 4, dropped: 0, current: 7.94, average: 7.94
last-message = rendered: 8, dropped: 0, current: 6.07, average: 6.88
last-message = rendered: 12, dropped: 0, current: 6.08, average: 6.59
last-message = rendered: 16, dropped: 0, current: 6.07, average: 6.45
```

4. Decode with VPP procamp
```
gst-launch-1.0 -vf filesrc location=test-input.h264 \
 ! h264parse ! vaapih264dec \
 ! vaapipostproc saturation=1.1 contrast=1.1 hue=2 \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1 display=drm" \
   sync=false text-overlay=false

last-message = rendered: 97, dropped: 0, current: 193.01, average: 193.01
last-message = rendered: 182, dropped: 0, current: 168.89, average: 180.94
last-message = rendered: 271, dropped: 0, current: 169.61, average: 177.06
last-message = rendered: 361, dropped: 0, current: 170.81, average: 175.46
```

5. Decode without VPP
```
gst-launch-1.0 -vf filesrc location=test-input.h264 \
 ! h264parse ! vaapih264dec \
 ! fpsdisplaysink video-sink="vaapisink fullscreen=1 display=drm" \
   sync=false text-overlay=false

last-message = rendered: 187, dropped: 0, current: 173.31, average: 186.61
last-message = rendered: 274, dropped: 0, current: 173.72, average: 182.31
last-message = rendered: 361, dropped: 0, current: 173.80, average: 180.19
last-message = rendered: 451, dropped: 0, current: 173.91, average: 178.90
```

uartie, 2020-04-20:
cc: @Jister @jbolduc01

uartie, 2020-04-20:
Similarly, I tried with msdk's `sample_vpp` composite and it reports a low 9 fps, too.

Jister, 2020-04-21:
Does such performance issue only exist on APL? How about other platforms like KBL or ICL?

uartie, 2020-04-21:
> Does such performance issue only exist on APL? How about other platforms like KBL or ICL?

As specified in the description (above), the issue does not appear to exist on KBL.  I have not tried any other platforms, except APL.

FurongZhang, 2020-04-21:
Yes, I know this issue and will check that.

jbolduc01, 2020-04-21:
Tried the setup on Coffee Lake platform and everything works fine.

FurongZhang, 2020-04-22:
This perf issue is specific to APL. This is an known issue. I am working on this. Actually, I had a fix, WIP to test the fix.

jbolduc01, 2020-05-11:
Is there any improvement on this issue yet?

FurongZhang, 2020-05-11:
We should not always use map gtt for APL.

jbolduc01, 2020-05-14:
Is there a way to test your fix on my side?

criusxu, 2021-06-03:
Is there any progress on this issue？
I  use this pipeline is ok on j3455 cpu.

gst-launch-1.0 vaapioverlay name=overlay min-upstream-latency=5000000000 \
 sink_0::xpos=0 sink_0::ypos=0  \
! vaapisink sync=false \
v4l2src  device=/dev/video0  ! "video/x-raw, width=1920, height=1080"  \
! vaapipostproc ! "video/x-raw(memory:VASurface),format=YV12,width=1920,height=1080" ! overlay.

but very slow use below pipeline.

gst-launch-1.0 vaapioverlay name=overlay min-upstream-latency=5000000000 \
 sink_0::xpos=0 sink_0::ypos=0  \
! vaapisink sync=false \
rtspsrc location=rtsp://192.168.0.179/ch1  latency =0 ! rtph264depay ! h264parse ! vaapih264dec \
! vaapipostproc ! "video/x-raw(memory:VASurface),format=YV12,width=1920,height=1080" ! overlay.

jbolduc01, 2021-06-03:
No progress. We dropped this platform…
Sorry

criusxu, 2021-06-03:
why the data come from v4l2src is ok?  the data all is  memory:VASurface.
Can I fix this issue by myself?

FurongZhang, 2021-06-03:
Please try this PR https://github.com/intel/media-driver/pull/646

criusxu, 2021-06-03:
> Please try this PR #646

It does not work for this issue. 
What else can I try　？

FurongZhang, 2022-11-20:
May I know if the issue is resolved or not. APL has been there for many years.

FurongZhang, 2022-11-20:
Please let me know if you still have issues. We can discuss here again.

closed 2022-11-20 by FurongZhang