intel/media-driver #916: Regression: image corruption since commit 1478047 when decoding a 4:4:4 JPEG file

xhaihao, 2020-04-23:
Reproduce steps:
1.  
```
$> gst-launch-1.0 filesrc location=input.444P.jpeg ! jpegparse ! vaapijpegdec ! filesink location=output.yuv
```

2. View output.yuv in a yuv viewer, the format is 444P. 

I see this issue on ICL and TGL

FurongZhang, 2020-04-23:
@XinfengZhang

Jister, 2020-04-24:
@xhaihao could you provide the test file? Thanks

xhaihao, 2020-04-24:
@Jister I have sent the clip to you by email

Jister, 2020-04-26:
Please try #920

xhaihao, 2020-04-27:
@Jister your patch works for me, thanks!

FocusLuo, 2020-04-27:
This patch also caused almost all of ffmpeg-vaapi and gst-vaapi vp9 10bit decode test cases failed, which dropped down big pass rate on us
can you help to fix it ASAP?

FocusLuo, 2020-04-27:
gst-launch-1.0 -vf filesrc location=./vp9_10bit_fastfood_64x64.ivf ! ivfparse ! vaapivp9dec ! videoconvert dither=0 ! video/x-raw,format=P010_10LE ! checksumsink2 file-checksum=false qos=false frame-checksum=false plane-checksum=false dump-output=true dump-location=test.yuv

Jister, 2020-04-27:
@FocusLuo could you share the test file './vp9_10bit_fastfood_64x64.ivf' , thanks

Jister, 2020-05-14:
@xhaihao Can we close the issue since the issue has been fixed? Thanks

closed 2020-05-17 by FurongZhang