intel/media-driver #918: [regression][HEVCe] VA_RC_QVBR unsupported (since 296ad0014649)

uartie, 2020-04-23:
VA_RC_QVBR is reported as unsupported for non-lowpower (VAEntrypointEncSlice) HEVC encode since:
```
commit 296ad00146499e6aecb247a2cb0fdef09b64bce1
Author: ShawnLi <shawn.li@intel.com>
Date:   Thu Apr 2 15:01:17 2020 +0800

    [Encode] fix rate control caps issue
    
    VAConfigAttribRateControl do not support VA_RC_QVBR at
    VAEntrypointEncSliceLP before gen12
    
    Change-Id: Ib1fe947169aff4324072b052c15425be7ce1bfa8
```

Based on the commit message, It appears the intention was to disable QVBR for lowpower (VAEntrypointEncSliceLP) HEVC encode ***only***.  However, it got removed for non-lowpower, too!

```
gst-launch-1.0 -vf videotestsrc num-buffers=5 \
  ! video/x-raw,format=NV12 \
  ! vaapih265enc rate-control=qvbr tune=0 \
  ! h265parse ! fakesink

ERROR vaapicontext gstvaapicontext.c:315:config_create: unsupported VA_RC_QVBR rate control
```

This regression was reproduced on KBL and ICL.  But it likely exists on all platforms before gen12.

Xiaogangli-intel, 2020-04-23:
Thank you for reporting this issue, will check it.

uartie, 2020-04-24:
PR already submitted #919

uartie, 2020-06-15:
@Xiaogang-Li @shawnli2 can you verify/review #919 please?

PR #919 (2020-07-02): [Encode] fix hevc QVBR BRC caps regression
VAConfigAttribRateControl supports VA_RC_QVBR at
VAEntrypointEncSlice.

This fixes a regression introduced by
```
 commit 296ad00146499e6aecb247a2cb0fdef09b64bce1
 Author: ShawnLi <shawn.li@intel.com>
 Date:   Thu Apr 2 15:01:17 2020 +0800

    [Encode] fix rate control caps issue

    VAConfigAttribRateControl do not support VA_RC_QVBR
    at VAEntrypointEncSliceLP before gen12
```
Fixes #918
commit 0c96976a (2020-07-02): [Encode] fix hevc QVBR BRC caps regression

VAConfigAttribRateControl supports VA_RC_QVBR at
VAEntrypointEncSlice.

This fixes a regression introduced by

 commit 296ad00146499e6aecb247a2cb0fdef09b64bce1
 Author: ShawnLi <shawn.li@intel.com>
 Date:   Thu Apr 2 15:01:17 2020 +0800

    [Encode] fix rate control caps issue

    VAConfigAttribRateControl do not support VA_RC_QVBR
    at VAEntrypointEncSliceLP before gen12

Fixes #918
closed 2020-07-02 by intel-mediadev