intel/media-driver #927: VPP can not enable more feature dynamically

HeJunyan, 2020-05-05:
Create a VPP using
vaCreateConfig(display, VAProfileNone, VAEntrypointVideoProc, NULL, 0, &config);
and create the context using
vaCreateContext (display, config, 0, 0, 0, NULL, 0, &context);

then, let the context do simple CSC job using vaBeginPicture/vaRenderPicture/vaEndPicture call. it can successfully generate output frames.
then, using the same context, enable some filter, such as `brightness`, while do the CSC job,  the VPP fails and report:

[MHW]:  CRITICAL - SetVeboxSurfaces:363: Unsupported format.
[MOS]:  CRITICAL - SubmitCommandBuffer:1183: Command buffer submission failed!
[VP]:   CRITICAL - VeboxSendVeboxCmd:2169: MOS returned error, eStatus = 0x22
[VP]:   CRITICAL - VeboxRenderMode0:3390: MOS returned error, eStatus = 0x22
[VP]:   CRITICAL - Render:3598: MOS returned error, eStatus = 0x22
[VP]:   CRITICAL - VpHal_RndrRenderVebox:4387: MOS returned error, eStatus = 0x22
[VP]:   CRITICAL - RenderSingleStream:850: MOS returned error, eStatus = 0x22

We found that, create a vpp context and do simple jobs for first several frames(may not enable VEbox) and then enable some complicated feature(may enable VEbox) later for the same context, this bug generates. 
But if we enable the complicated feature at the first time(the first frame), it's OK.

FurongZhang, 2020-05-06:
@HeJunyan  What kinds of format you are checking? The above log shows that VEBOX can not support the format. I think CSC could be done in VEBOX or EU. Obviously, in your experiment, CSC could be done in EU.

HeJunyan, 2020-05-06:
For example, create a VPP context, let first 100 frames just do CSC(nv12->RGBA), and this does not need VEBox. Then keep the same context,  let another 100 frames do CSCCSC(nv12->RGBA)+brightness, then this bug generates, report no VEbox

FurongZhang, 2020-05-06:
@HeJunyan  Do you have a test app so that I can reproduce it?

FurongZhang, 2020-05-06:
Or you would like to file a PR to fix it? It is up to you.

HeJunyan, 2020-05-06:
We find it in gstreamer.  We convert the decoder's output NV12 to RGBA to glimgesink. User change the brightness at the play time and this bug generates.
I attached the test file.

You can also reproduce it by any CSC+brightness(or contrast ...) test case, and disable the  brightness for the first few frames and then enable it.

HeJunyan, 2020-05-06:
[hello_gst.c.zip](https://github.com/intel/media-driver/files/4585348/hello_gst.c.zip)

MicroYY, 2022-01-18:
I'm trying to reproduce it.