intel/media-driver #930: [Issue] KO hardware decode/encode with Gemini lake family

Jqh63, 2020-05-10:
## Desc: Hardware media decode/encode is KO with iHD driver on Gemini lake family
Unable to perform media decode/encode with iHD_drv_video.so, always have to force use of i965 driver. Impacts are degraded performance and capabilites: no HEVC 10bits for example or high CPU usage. Plus manual configuration is needed, like removal of iHD_drv_video.so or set up LIBVA_DRIVER_NAME to i965

## Current behavior
iHD_drv_video.so generate error message or hangs on Gemini lake processors

## Expected behavior
Working hardware decode as mention into VAAPI and Intel spec for GLK (especially HEVC 10bits HW encode)

## How to reproduce
1. run a Debian/Ubuntu/Alpine (docker or direct host)
2. install intel-media-driver package to get a working VAAPI
3. run a HW encode (plex, handbrake, ...), simple way is a handbrake test with H.265 Intel QSV (docker compose below)
4. CPU/GPU hangs, artifacts into generated files, error messages..
```
[16:42:39] hb_display_init: using VA driver 'iHD'
libva info: VA-API version 1.7.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /opt/intel/mediasdk/lib64/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_7
libva info: va_openDriver() returns 0
[16:42:40] qsv_hevc_make_header: MFXVideoCORE_SyncOperation failed (-17)
[16:42:40] encqsvInit: qsv_hevc_make_header failed
[16:42:40] Failure to initialise thread 'Quick Sync Video encoder (Intel Media SDK)'
```

## Ecosystem
On my test case:
- Asrock j5005 
- running OMV5 (Debian 10)
- with docker and multiple images like LSIO Plex, Handbrake, ...

## Compose
```
version: '2.1'
services:
  handbrake:
    image: jlesage/handbrake
    container_name: handbrake
    environment:
      - USER_ID=1000
      - GROUP_ID=100
      - TZ=Europe/Paris
      #- LIBVA_DRIVER_NAME=i965
    ports:
      - 5800:5800
    volumes:
      - /srv/dev-disk-by-label-data/appdata/handbrake:/config:rw
      - /srv/dev-disk-by-label-media/media:/storage:ro
      - /srv/dev-disk-by-label-media/media/handbrake/watch:/watch:rw
      - /srv/dev-disk-by-label-media/media/handbrake/output:/output:rw
    devices:
      - /dev/dri:/dev/dri
    restart: unless-stopped
```

## Refs 
[plex forum - tons of users issues](https://forums.plex.tv/t/corrupted-video-output-stream-with-hardware-transcoding/500432/86)
[plex docker image](https://github.com/linuxserver/docker-plex/issues/228)
[maybe related to issue 733](https://github.com/intel/media-driver/issues/733)

mrzippyuk, 2020-06-06:
Any update on this, or additional info required as I also see the same issue here too.

Using Gigabyte Brix 4105C - so the J4105 processor.

Ubuntu 20.04.

NeV3rKilL, 2020-06-26:
Same behavior here with another:

model name      : Intel(R) Celeron(R) J4105

argakon, 2020-07-01:
Same problem with:

MB: Asrock J4105-ITX

Artifacts in encoded video file.

THX723, 2020-07-13:
Adding issue to include all new 2020 Synology NAS based on J4105 and J4125 (GLK) CPU.
Observing heavy artifact issue, especially with HEVC 8-bit.

XinfengZhang, 2020-08-07:
#1005 fixed basic function --- memory access of GLK. actually, because i965 and iHD both cover gen9 platforms,   we suggest to switch driver from ICL.

Rimasi89, 2020-08-13:
Same issue even with the new updated plex apk (PlexMediaServer-1.19.5.3112)

NAS: Asustor 5202t
CPU: Intel Celeron J4005

iHD.bak workaround was still needed :/

tlonic, 2020-10-02:
Issue still present in Q3.2020 release:

CPU: Celeron J4115

```
[13:45:23] hb_display_init: using VA driver 'iHD'
libva info: VA-API version 1.9.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_9
libva info: va_openDriver() returns 0
[13:45:23] encqsvInit: MFXVideoENCODE_Init failed (-17)
[13:45:23] Failure to initialise thread 'Quick Sync Video encoder (Intel Media SDK)'
```

Rimasi89, 2020-10-02:
I can conclude that this fix is not on the top priority list from both Plex and Intel. Nice one! 👏

DOFfactory, 2020-10-06:
Well, more and more NASes are going to reach the market in the coming months with j4125 Celeron CPUs. The only way is to keep reminding Intel I guess...

My case is Asustor AS6604T unable to hardware transcode even into H264, while much older processors were completely fine.

l1ge, 2020-10-09:
I just got the new Synology DS1520+ which uses a J4125 Celeron, so i'd love a fix too for this driver!

davidsonb, 2020-10-09:
Same problem. Plex and Synology DS1520+ which uses a J4125 Celeron.

jara1, 2020-10-09:
Same problem Qnap TS453D - Celeron J4125.

Rimasi89, 2020-10-09:
> Same problem. Plex and Synology DS1520+ which uses a J4125 Celeron.

You should wait sitting............ They just don't even care. Since you pay every month they're happy...

DOFfactory, 2020-10-09:
Shaming and naming ... they will care eventually :D

Rimasi89, 2020-10-09:
> Shaming and naming ... they will care eventually :D

I wouldn't be so sure about that. With answers like this one bellow reflects how they care about they're customers and how about they care about the new one to come:

[image]

👏👏👏

PLEX: "Every user has to decide for themselves whether they wish to continue a Plex Pass subscription. We can't make that decision for a user, of course."

👏👏👏

DOFfactory, 2020-10-09:
Oh, I meant Intel and not Plex. I think the issue is with the driver and not with the Plex as an backend. I thought this is about Intel...

l1ge, 2020-10-09:
yep i don't think Plex are the ones to blame now.

Xiaogangli-intel, 2020-10-10:
Today, I checked #733, @XinfengZhang's PR #1005 should already fix this issue.
I think these 2 issues should have the same root cause.
Could you try the latest source and try again?

XinfengZhang, 2020-10-10:
1. have anyone tried #1005?? which already been merged. it should already fix most of GeminiLake issue. because this issue is mixed with MSDK issue https://github.com/Intel-Media-SDK/MediaSDK/issues/1663 , please also make sure msdk include https://github.com/Intel-Media-SDK/MediaSDK/pull/1770
2. another problem is about application, as I know, https://github.com/intel/intel-vaapi-driver support GLK, to keep the support smooth, we recommend that users who are using i965 driver continue to use it on GLK.  because the feature set maybe is different , for example: some VPP algorithm is different.
of course, this case maybe is special? because Plex is using MSDK, MSDK works good with media_driver.
but for most of usage. Application installation could choose different UMD on different platform. right?

l1ge, 2020-10-10:
I would like to try if its solved with the latest source. But I have no idea how to replace the current driver version I have on my Synology DS1520+ by the new version that you fixed. Is it possible to do it by myself or do we have to wait for Synology to update it themselves?

tlonic, 2020-10-10:
As seen in [my comment above](https://github.com/intel/media-driver/issues/930#issuecomment-702469826), at least HEVC 10-bit encoding is still broken using bleeding edge packages.

tlonic, 2020-10-11:
HandBrakeCLI hangs at 0% when trying to encode using the QSV decoder with a HEVC 10-bit file, again, using bleeding edge packages: 

kernel: 5.8.13-arch1-1
intel-gmmlib-git: 20.3.2.r0.g2072b0d-1
intel-media-driver-git: 2020.3.pre8.r94.g8de44c97-1
intel-media-sdk-git: 2020.3.pre6.r45.g7beb403e-1
CPU: Celeron J4115

`x264 [info]: using SAR=1/1
x264 [info]: using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2
x264 [info]: profile High, level 4.0, 4:2:0, 8-bit
Encoding: task 1 of 1, 0.00 %[13:04:21] sync: first pts audio 0x1 is 0
Encoding: task 1 of 1, 0.00 %^CSignal 2 received, terminating - do it again in case it gets stuck
Encoding: task 1 of 1, 0.00 %[13:06:32] work: average encoding speed for job is 0.000000 fps
[13:06:32] aac-decoder done: 73073 frames, 0 decoder errors
[13:06:32] hevc_qsv-decoder done: 0 frames, 37374 decoder errors
[13:06:32] sync: got 0 frames, 120941 expected
[13:06:32] mux: track 0, 0 frames, 0 bytes, 0.00 kbps, fifo 8
[13:06:32] mux: track 1, 72873 frames, 24288647 bytes, 124.99 kbps, fifo 131072
[13:06:32] Finished work at: Sat Oct 10 13:06:32 2020

HandBrake has exited.
`

XinfengZhang, 2020-10-12:
> I would like to try if its solved with the latest source. But I have no idea how to replace the current driver version I have on my Synology DS1520+ by the new version that you fixed. I'm using plex for the transcode. Anyone has an idea?

replace iHD package or redirect it by such env variables :LIBVA_DRIVERS_PATH=xxx

Xiaogangli-intel, 2020-10-13:
@tlonic I checked with sample_decode, decode can work properly, you need to confirm whether your input stream is 10-bit HEVC. If yes, please provide your command line and stream. Thank you.

For HEVC 10-bit encode,  I did find an issue on GLK, will fix it ASAP.

Jqh63, 2020-11-29:
@XinfengZhang Hello, any update about merge request? how can I test it anyway

Freeben666, 2020-12-01:
Any news on this ? I have an Asustor AS6604T NAS with a Celeron J4125 that is impacted.

k29t59dh, 2021-01-04:
Has anyone successfully encoded **8-bit** H.265 on Gemini Lake with this fix? 
10 bit H.265 works on my j5005 but 8-bit fails.

XinfengZhang, 2021-01-05:
the issue was closed automatically, because #1078 mentioned it fixed this issue. @Xiaogang-Li , we should already test 8 bit hevc encoding, it is strange , 10bit works  ,but 8 bit failed.  @k29t59dh , could you help to add more details?

k29t59dh, 2021-01-06:
I'm getting the error in a HandBrake docker with driver version 20.4.4.
What other details would be helpful - any commands I can run (and paste the output) or logs I should include?

jotacor, 2021-01-06:
It happens the same to me with version 20.4.5.
It encodes fine 10 bit x265 and x264, but not 8 bits.
I'm working with an Intel j4125 processor.

Rimasi89, 2021-01-06:
I never got it fixed. I've disable the iHD intel driver!

I got an ASUSTOR NIMBUSTOR 2 with the intel J4005 and the issue was never fixed. I dont know why you closed the thread.

Obter o Outlook para Android<https://aka.ms/ghei36>

XinfengZhang, 2021-01-07:
@Rimasi89 , have you tried the patch? if it was not fixed. let's reopen it, @Xiaogang-Li , could you help to double confirm whether it already been fixed?

jotacor, 2021-01-07:
where/what is the patch?

Rimasi89, 2021-01-07:
I couldn't see the Patch like @jotacor

Obter o Outlook para Android<https://aka.ms/ghei36>

esmorun, 2021-02-01:
I still can't do hardware transcoding in Jellyfin because of this issue. I'm on Ubuntu 20.04LTS and an Asrock j5005-board. Intel-media-va-driver-non-free (20.1.1+ds1-1build1).

If there is a patch, please let me know how I can test it and report back to you.

mkarwin, 2021-02-22:
Has the issue been fixed on the NAS devices? Synology DSXX20+, Qnap TS-X53D, Asustor AS660XT and their rackmount counterparts are all affected it seems. Any owners care to verify that their platform's working fine already or to explicitly verify that the issue persists?
The Gemini Lake Refresh CPUs are more or less ubiquituous amongst the NAS appliances from most NAS manufacturers that have been released within the last two years. It's almost 9 and a half months since the issue was first reported. Intel surely knows that these CPUs have been sold in large quantities to NAS manufacturers for home type appliances where transcoding/encoding/decoding features are of paramount importance since there is no easy way to replace said CPUs are gain additional peformance through add-in cards. As such, this should have already been fixed and verified with NAS manufacturers as parts of their respective OS updates.

DonCaliOni, 2021-03-30:
> Has the issue been fixed on the NAS devices? Synology DSXX20+, Qnap TS-X53D, Asustor AS660XT and their rackmount counterparts are all affected it seems. Any owners care to verify that their [...]

Hi, I have a QNAP TS-453D and I'm still having this damn issue!

DonCaliOni, 2021-03-30:
> Same problem Qnap TS453D - Celeron J4125.

Hi, did you ever get this sorted? I have the same NAS and still experiencing the issue...

DOFfactory, 2021-03-30:
Here is a solution that works for Asustor (check YeJ's comment): https://emby.media/community/index.php?/topic/89948-hardware-encoding-issues-j4125/

Basically you have to delete/move the `iHD_drv_video.so` file from the libraries, and keep the `i965_drv_video.so` there. The hw with this driver seem to work.

jotacor, 2021-03-30:
> Here is a solution that works for Asustor (check YeJ's comment): https://emby.media/community/index.php?/topic/89948-hardware-encoding-issues-j4125/
>
> Basically you have to delete/move the [...]

That is a workaround, not a solution.
A solution would be the developers fixing the issue affecting ALL Gemini Lake Intel processors.

DOFfactory, 2021-03-30:
You are right, apologies for the bad wording, it is a workaround not a solution. 

I would also welcome from @Xiaogang-Li (assigned intel dev) an update on progress...

venzislav, 2021-06-06:
Please fix it https://github.com/Xiaogang-Li :)

Xiaogangli-intel, 2021-07-01:
#1215 this will fix the 8-bit encoding issue for HEVC. If you met this issue, please try this PR.
And 10-bit encoding should work.
If you still meet encoding issue on GLK, please send out your command line, then I will check it.
Thank you!

Xiaogangli-intel, 2021-07-19:
#1215 is merged, I will close this issue in two days if no more questions. Please file a new one if you still meet other issues of GLK.
Thank you.

Jqh63, 2021-10-31:
Seems OK now on last LSIO Plex from 4K HDR H265 to 1080p H264 using HW, nice!

BjoernUsw, 2022-02-07:
Just checked with the latest Version of Plex for DSM6 on an DS920+ (Plex 1.25.4.5487). Trying to transcode 1080p H265 to 1080p H264 to burn in ASS subtitles for streaming to android devices still buffers every 5 seconds for at least 20 seconds using the `iHD_drv_video.so` but works fine with the `i965_drv_video.so`.

Xiaogangli-intel, 2022-02-09:
@BjoernUsw could you provide some logs and dmesg result? Thank you.

ofsaleem, 2022-02-13:
> Just checked with the latest Version of Plex for DSM6 on an DS920+ (Plex 1.25.4.5487). Trying to transcode 1080p H265 to 1080p H264 to burn in ASS subtitles for streaming to android devices still [...]

i can confirm im having this problem on a qnap ts-653D with plex to *any* client that isnt directplaying, even for 2mbps 720p files. putting `VaapiDriver="i965"` at the end of the Preferences.xml file in my server folder completely solved the issue

robertofabrizi, 2022-03-28:
Almost two years and Intel hasn't fixed this yet?

Xiaogangli-intel, 2022-03-29:
This issue should be fixed. If you still meet this issue, please try the latest driver. If still cannot solve issue by updating driver, please provide your environment detail(app/driver/KMD version and logs). Thank you.

DonCaliOni, 2022-04-09:
> This issue should be fixed. If you still meet this issue, please try the latest driver. If still cannot solve issue by updating driver, please provide your environment detail(app/driver/KMD version and logs). Thank you.

How do I go about updating the driver on a QNAP NAS? I've had a Google but it's not been much help. Thanks

DonCaliOni, 2022-04-09:
> > Just checked with the latest Version of Plex for DSM6 on an DS920+ (Plex 1.25.4.5487). Trying to transcode 1080p H265 to 1080p H264 to burn in ASS subtitles for streaming to android devices still buffers every 5 seconds for at least 20 seconds using the `iHD_drv_video.so` but works fine with the `i965_drv_video.so`.
> 
> i can confirm im having this problem on a qnap ts-653D with plex to _any_ client that isnt directplaying, even for 2mbps 720p files. putting `VaapiDriver="i965"` at the end of the Preferences.xml file in my server folder completely solved the issue
> 
> i don't really understand how to tell what version of the ihd driver i have or how to upgrade it if i need to, to make sure im using a version with the fix. im new to NASes and all the online information is about synology devices, and i can't figure out how to see driver version information on the qnap OS.

Hi, did you manage to upgrade the driver? I have the same NAS (just the 4 bay TS-453D) and haven't figured out how to update the driver...

DOFfactory, 2022-04-09:
I think you need to drop an email to the support of the vendor, as the driver usually comes with the firmware update.

tekwarfare, 2022-08-24:
> Just checked with the latest Version of Plex for DSM6 on an DS920+ (Plex 1.25.4.5487). Trying to transcode 1080p H265 to 1080p H264 to burn in ASS subtitles for streaming to android devices still [...]

I'm also having this exact issue. I'm running Unraid 6.10.3 on an Asrock J5005 with Plex Media Server 1.28.2.6106. I still have to delete iHD_drv_video.so as a workaround. 

Do you have to build on the OS you plan to use the driver? Can I compile this under VMware and then copy the driver to my server? This stuff is beyond me and I can never get things like this to work so I would be eternally grateful if someone could build the latest version of iHD_drv_video.so

ofsaleem, 2023-10-22:
hi, i'm still awaiting instructions/guidlines from anyone on how to determine what **version** of the driver i'm using to determine if i'm using the one with the fix in it

DonCaliOni, 2023-10-22:
> hi, i'm still awaiting instructions/guidlines from anyone on how to determine what **version** of the driver i'm using to determine if i'm using the one with the fix in it

At this point, unless you're actually having buffering issue only with HEVC, assume it has been fixed...

ofsaleem, 2023-10-22:
> > hi, i'm still awaiting instructions/guidlines from anyone on how to determine what **version** of the driver i'm using to determine if i'm using the one with the fix in it
> 
> At this point, unless you're actually having buffering issue only with HEVC, assume it has been fixed...

apologies, i'm dealing with concurrent issues and didn't want to adjust my configuration (and potentially crash my server again) without confirmation the issue should have disappeared. i tested anyway and it seems to be working as far as i can tell. on a qnap-ts653d

PR #1078 (2020-12-03): Fix HEVC encoding failed on P frame issue on GLK
Together with  #1005 can fix following 2 issues. Fixed #930, Fixed #733.
commit 91baad76 (2020-12-03): Fix HEVC encoding failed on P frame issue on GLK

Together with  #1005 can fix following 2 issues. Fixed #930, Fixed #733.