intel/media-driver #935: deb package for media driver?

ambicapramod, 2020-05-19:
I'm able to build and install the driver from the instructions given at https://github.com/intel/media-driver/wiki/How-to-build-and-use-media-driver%3F ;

But I cannot build everytime to install the driver on other ubuntu platform. Is there any way I can put it in deb pkg or anyother sort of all the dependencies (libva,gmmlib) etc and install by dpkg -i ? 

I could not find the instructions for the same.

XinfengZhang, 2020-05-25:
cat << EOF | sudo tee /etc/apt/sources.list.d/intel-graphics.list
deb [trusted=yes arch=amd64] https://repositories.intel.com/graphics/ubuntu eoan main
EOF
sudo apt-get update

then sudo apt-get install intel-media-va-driver-non-free

pythonsan, 2020-07-10:
@XinfengZhang I encountered a new problem, follow your guide :-(

```
root@demo: sudo apt-get install -s intel-media-va-driver-non-free
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 intel-media-va-driver-non-free : Depends: libc6 (>= 2.29) but 2.27-3ubuntu1.2 is to be installed
E: Unable to correct problems, you have held broken packages.

18.04.4 LTS (Bionic Beaver)
5.3.0-62-generic #56~18.04.1-Ubuntu SMP Wed Jun 24 16:17:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

Upgrading libc is a risky operation, so can it support ubuntu 18.04 of intel-graphics.list ?

XinfengZhang, 2020-07-11:
it should support ubuntu 18.04. I suppose , it is a package problem. they build the package on 19.04 or 20.04.  will check it with maintainer of the package.

dvrogozh, 2020-07-11:
@XinfengZhang : no, https://repositories.intel.com/graphics/ubuntu support only Ubuntu 19.10 (eoan). Ubuntu 20.04 (focal) is in plans, but I think packages are not there yet. I don't think that the named repository supports packages for Ubuntu 18.04 (bionic) at least for media.

@ambicapramod : since you are on Ubuntu 18.04, but try to install packages from Ubuntu 19.10, you step into dependency issues. Basically you will upgrade to the later ubuntu version if you will peruse this path which might not be what you actually want to do. My recommendation however would be to consider permanent update to Ubuntu 20.04 at the first place and use intel-media-va-driver-non-free out of the stock Ubuntu (it appeared in Ubuntu >= 19.04). If you will need more recent media driver version, then you can consider using https://repositories.intel.com/graphics/ubuntu going forward since current ubuntu stock package version is eventually pretty recent).

pythonsan, 2020-07-12:
Thank you very much for your reply

Because we use specific hardware, its operation must depend on the 18.04 version, so we cannot upgrade to a higher ubuntu temporarily.

I have given up installing media-driver through apt-get. I have solved this problem using the source code compilation and installation mode. If 18.04 is supported in the future, please tell me, thank you :-)

@XinfengZhang @dvrogozh

XinfengZhang, 2020-07-13:
@ambicapramod  , sorry, I dont know your OS version before I give your command line, for 8.04, you should use this one (but the driver maybe is not latest one):  cat << EOF | sudo tee /etc/apt/sources.list.d/intel-graphics.list
deb [trusted=yes arch=amd64] https://repositories.intel.com/graphics/ubuntu bionic main
EOF

XinfengZhang, 2020-08-07:
@ambicapramod  I will close the issue, if you have any other concern, please add comment here, we could re-open it

closed 2020-08-07 by XinfengZhang