intel/media-driver #937: mediadriver build fails on 32 bit

bosepriy, 2020-05-21:
Hi,

I am trying to build the media driver for 32 bit platform using  -DARCH=32 on 16.04.1-Ubuntu
I am getting errors :
====
media-driver/media_driver/linux/common/os/i915/include/libdrm_macros.h:40:40: error: size of array is negative
       (void) sizeof(char [1 - 2*!(COND)]); \         
                                        ^
/home/bosepriy/Downloads/libva/media_driver/media-driver/media_driver/linux/common/os/i915/include/libdrm_macros.h:76:4: note: in expansion of macro ‘STATIC_ASSERT’
    STATIC_ASSERT(LARGE_OFF_T % 2147483629 == 721 &&
========

Is there some patch that we can use to make a build of media driver on 32bit platforms ?

XinfengZhang, 2020-05-25:
@ljia5, could you help to take a look?

XinfengZhang, 2020-05-26:
@bosepriy it is duplicate with #356

XinfengZhang, 2020-08-07:
copied from #356 need to enable a special CXXFLAGS such as -D_FILE_OFFSET_BITS=64 or by default the value is 32.

closed 2020-08-07 by XinfengZhang