intel/media-driver #941: Need GRAY8 format to handle YUV400 decode

HeJunyan, 2020-05-26:
when chroma_format_idc is 0 in h264, it is a pure gray8 video format. And we should use GRAY8 surface as the RTFormat to decode this kind of stream. The current still use NV12 and set the UV plane to ignored value, which is not very good. According to @xhaihao , the HW supports the GRAY8 format and so should improve this.

wangyan-intel, 2020-06-08:
@weizhu-intel Could you please check it? Thanks.

weizhu-intel, 2020-06-09:
Yan, can you check which platform they are working on?

wangyan-intel, 2020-06-09:
@HeJunyan What platform are your working? TGL?

HeJunyan, 2020-06-09:
I test it on SKL/ICL. But seems all platform use NV12, no gray8 available.

weizhu-intel, 2020-08-07:
Our HW can't use yuv400 directly.  If give yuv400,  we need to convert to nv12 in driver.

closed 2020-08-07 by XinfengZhang