intel/media-driver #947: Internal decoding error (23) - Kodi 18.6, Ubuntu 20.04, j4105 UHD600

teeedubb, 2020-05-27:
Internal decoding error (23) - Kodi 18.6, Ubuntu 20.04, j4105 UHD600
System information

System: Host: blaster Kernel: 5.4.0-31-generic x86_64 bits: 64 compiler: gcc v: 9.3.0 Desktop: N/A dm: LightDM Distro: Ubuntu 20.04 LTS (Focal Fossa) CPU: Topology: Quad Core model: Intel Celeron J4105 bits: 64 type: MCP arch: Goldmont Plus rev: 1 L2 cache: 4096 KiB flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 11980 Speed: 768 MHz min/max: 800/2500 MHz Core speeds (MHz): 1: 960 2: 778 3: 853 4: 1072 Graphics: Device-1: Intel UHD Graphics 605 driver: i915 v: kernel bus ID: 00:02.0 chip ID: 8086:3185 Display: server: X.Org 1.20.8 driver: intel resolution: 1920x1080~60Hz, 1920x1080~60Hz OpenGL: renderer: Mesa Intel UHD Graphics 600 (GLK 2) v: 4.6 Mesa 20.0.4 direct render: Yes

Describe the issue
Random black flashes lasting about less than a second appear onscreen when playing back video with Kodi 18.7 on Ubuntu 20.04. When the flashes occur I can see the following error in the kodi log file - "ERROR: CVaapi2Texture::Map: vaSyncSurface - Error: internal decoding error (23)". Vaapi is in use and the media is x264 content. I never had this issue with Ubuntu 19.10 (it was a in place upgrade). Adding the LIBVA_DRIVER_NAME=i965 environment variable to /etc/environment fixes the black flashes for me. Using just MESA_LOADER_DRIVER_OVERRIDE=i965 the flashes still persist. If I use both environment variables the flashes stop.

Regression
Worked fine in Ubuntu 19.10

Log files as attachment
dpkg -l |grep mesa | pastebinit https://paste.ubuntu.com/p/GKnkhbpkT7/

DISPLAY=:0 vainfo | pastebinit libva info: VA-API version 1.7.0 libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so libva info: Found init function __vaDriverInit_1_7 libva info: va_openDriver() returns 0 https://paste.ubuntu.com/p/vTT899SHDf/

cat ~/.kodi/temp/kodi.log | pastebinit https://paste.ubuntu.com/p/DzRqWZ7DSJ/

dmesg | pastebinit https://paste.ubuntu.com/p/dtvHKsXy97/

id | pastebinit https://paste.ubuntu.com/p/DRwM7syxyp/

amixer | pastebinit https://paste.ubuntu.com/p/pg2NmXpz7K/

cat /var/log/Xorg.0.log | pastebinit https://paste.ubuntu.com/p/5S9px8VSZG/

Original forum thread: https://forum.kodi.tv/showthread.php?tid=354755

XinfengZhang, 2020-06-08:
thanks reporting, the issue happens on Ubuntu20.4 , but not 19.04, same libva version and media driver version?

teeedubb, 2020-06-10:
I am not sure what the libva and media driver versions were in 19.10. I did not have to use the LIBVA_DRIVER_NAME=i965 environment variable in 19.10.

argakon, 2020-07-14:
I think this problem can be related to this #930

XinfengZhang, 2020-10-19:
if it is same as #930, it should already been fixed by #1005

XinfengZhang, 2021-01-08:
it should already been fixed by #1005, so I close it, any concern, please re-open it

closed 2021-01-08 by XinfengZhang