intel/media-driver #958: Compared to ICL, CSC on TGL has low quality.

xhaihao, 2020-06-04:
Reproduce step (e.g. convert 256x256 nv12 file to 128x128 nv12 file, then convert 128x128 nv12 file to 256x256 i420 file)
1. gst-launch-1.0 filesrc location=input.nv12.yuv ! rawvideoparse format=nv12 width=256 height=256 ! vaapipostproc format=nv12 width=128 height=128 ! filesink location=tmp.nv12.yuv

2. gst-launch-1.0 filesrc location=tmp.nv12.yuv ! rawvideoparse format=nv12 width=128 height=128 ! vaapipostproc format=i420 width=256 height=256 ! filesink location=output.i420.yuv

3. Calculating the psnr between input.nv12.yuv and output.i420.yuv. Compared to ICL, the result on TGL has 1db~2db gap. 

media-driver: d7061b608ad4d55fccab00f8d80b852f20eab1d7
gmmlib: d7a0586104096f3e2241e2a773c6bc41a9e2c422
gstreamer-vaapi: e41c6bd29895156cfdc4c36bd23826e9e3ac8a67

XinfengZhang, 2020-06-08:
@Jister @FurongZhang  could you help to take a look?

FurongZhang, 2020-06-08:
@Jister  to have a look.

xhaihao, 2020-07-03:
Any progress for this issue?

FurongZhang, 2020-09-15:
@Jister to update.

Jister, 2020-09-16:
@xhaihao For this case, first step is downscaling from NV12 to NV12, seconde step is upscaling from NV12 to I420. The downscaling is done by SFC, the upscaling is done by Render(since SFC cannot support I420). On TGL SFC scaling will enable output centering, but ICL SFC don't support, and alsot Render don't not support output centering on both TGL and ICL, which cause the difference.
One solution to avoid such difference, is the do upscaling and CSC separatly, i.e. NV12 to NV12 downscaling, NV12 to NV12 upscaling, NV12 to I420 CSC, there will be no such big PSNR drop.

XinfengZhang, 2021-01-08:
@xhaihao , does the explanation from Jason make sense, if so, please help to close it

MicroYY, 2021-11-11:
Hi @xhaihao 
This issue should be expected because output centering is introduced in TGL. Does the explanation make sense for you? 
Could you help close this issue?

> @xhaihao For this case, first step is downscaling from NV12 to NV12, seconde step is upscaling from NV12 to I420. The downscaling is done by SFC, the upscaling is done by Render(since SFC cannot support I420). On TGL SFC scaling will enable output centering, but ICL SFC don't support, and also Render don't support output centering on both TGL and ICL, which cause the difference. One solution to avoid such difference, is to do upscaling and CSC separatly, i.e. NV12 to NV12 downscaling, NV12 to NV12 upscaling, NV12 to I420 CSC, there will be no such big PSNR drop.

closed 2021-11-23 by MicroYY