intel/media-driver #973: [Encode] Differences in slice header prefix for AVC Vdenc and VME

ss22ongithub, 2020-06-15:
We tried to record a video using Android camera  application on the ChromeOS  (via the android VM ) using H264Main profile and EncSliceLP entrypoint. 

After the frame is encoded the App tries to validate certain length checks ,  the before writing to the disk  and this causes failure. 
Point of failure in android https://cs.android.com/android/platform/superproject/+/master:frameworks/av/media/libstagefright/MPEG4Writer.cpp;drc=f9d5d46591bb2ff31ce646dfa61911dfef5ac62e;l=1452

Observations from initial debug . 
-	App crash always occurs when LP encoder EncSliceLP is used , observed on KBL (eve) and JSL+.   Using the IHD Tip of  the tree for this .
-	The App crash is not seen when i965 driver is used in the KBL  with low power encode enabled 
-	There are difference in the  NAL bytes size , for iHD and i965 in LP mode . Please see the logs from the android side below .

Further debug from , 
Currently in case of Vdenc avc slice header prefix is 0x000001 (3bytes)
and in case of VME it is 0x00000001 (4 bytes) , so change slice header
prefix to 4 bytes for vdenc (same as i965 driver).

ss22ongithub, 2020-06-15:
A patch has been pushed here #971 for review

mattst88, 2022-09-06:
Please reopen. We're still carrying the patch in #971 in ChromeOS.