intel/media-driver #978: gpu hang when we do 1 to N transcode.

xuguangxin, 2020-06-17:
race parport_pc ppdev sunrpc lp parport ip_tables x_tables autofs4 hid_generic usbhid hid e1000e ptp pps_core ahci libahci
[ 1107.361254] CPU: 8 PID: 13462 Comm: queue3:src Not tainted 4.15.0-106-generic #107-Ubuntu
[ 1107.361255] Hardware name: Gigabyte Technology Co., Ltd. Z370 AORUS Gaming 5/Z370 AORUS Gaming 5-CF, BIOS F7 07/05/2018
[ 1107.361283] RIP: 0010:gen8_ppgtt_set_pde.isra.27+0x45/0x60 [i915]
[ 1107.361284] RSP: 0018:ffffbc75078778c8 EFLAGS: 00010206
[ 1107.361286] RAX: 84b18ff223e8e0d0 RBX: 0000000003400000 RCX: ffff9ac7e92b4680
[ 1107.361288] RDX: 00000001e908a003 RSI: ffff9ac832719f40 RDI: ffff9ac662200000
[ 1107.361289] RBP: ffffbc75078778c8 R08: 0000000000000000 R09: 0000000000000000
[ 1107.361290] R10: 0000000000000000 R11: 0000000000000000 R12: 000000000000001a
[ 1107.361291] R13: 0000000000070000 R14: ffff9ac662200000 R15: ffff9ac7ce582000
[ 1107.361293] FS:  00007ffb7b7fe700(0000) GS:ffff9ac836c00000(0000) knlGS:0000000000000000
[ 1107.361295] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1107.361296] CR2: 00007ff73df0a0a8 CR3: 0000000218a3e006 CR4: 00000000003606e0
[ 1107.361297] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 1107.361298] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 1107.361299] Call Trace:
[ 1107.361323]  gen8_ppgtt_alloc_pdp+0x1c2/0x340 [i915]
[ 1107.361343]  gen8_ppgtt_alloc_4lvl+0x56/0x150 [i915]
[ 1107.361362]  ppgtt_bind_vma+0x39/0x70 [i915]
[ 1107.361383]  i915_vma_bind+0x75/0xd0 [i915]
[ 1107.361402]  __i915_vma_do_pin+0x2ce/0x480 [i915]
[ 1107.361415]  eb_lookup_vmas+0x7b0/0xb40 [i915]
[ 1107.361418]  ? _cond_resched+0x19/0x40
[ 1107.361420]  ? __pm_runtime_resume+0x5b/0x80
[ 1107.361433]  i915_gem_do_execbuffer+0x4a2/0x1230 [i915]
[ 1107.361436]  ? get_page_from_freelist+0xf16/0x1400
[ 1107.361438]  ? shmem_getpage_gfp+0xe0/0xbd0
[ 1107.361440]  ? __kmalloc_track_caller+0x181/0x200
[ 1107.361442]  ? krealloc+0x70/0xd0
[ 1107.361444]  ? reservation_object_get_fences_rcu+0x28e/0x2b0
[ 1107.361458]  i915_gem_execbuffer2+0x1b8/0x390 [i915]
[ 1107.361473]  ? i915_gem_set_domain_ioctl+0x1a8/0x280 [i915]
[ 1107.361485]  ? i915_gem_execbuffer+0x2d0/0x2d0 [i915]
[ 1107.361494]  drm_ioctl_kernel+0x5f/0xb0 [drm]
[ 1107.361500]  drm_ioctl+0x38e/0x460 [drm]
[ 1107.361512]  ? i915_gem_execbuffer+0x2d0/0x2d0 [i915]
[ 1107.361515]  ? dput+0x13/0x20
[ 1107.361516]  ? _cond_resched+0x19/0x40
[ 1107.361517]  do_vfs_ioctl+0xa8/0x630
[ 1107.361519]  ? __fput+0x193/0x220
[ 1107.361521]  SyS_ioctl+0x79/0x90
[ 1107.361523]  do_syscall_64+0x73/0x130
[ 1107.361525]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[ 1107.361526] RIP: 0033:0x7ffb9b4aa5d7
[ 1107.361527] RSP: 002b:00007ffb7b7fccf8 EFLAGS: 00000206 ORIG_RAX: 0000000000000010
[ 1107.361528] RAX: ffffffffffffffda RBX: 00007ffb6c0ab790 RCX: 00007ffb9b4aa5d7
[ 1107.361529] RDX: 00007ffb7b7fcdc0 RSI: 00000000c0406469 RDI: 0000000000000005
[ 1107.361530] RBP: 00007ffb7b7fcd30 R08: 000055984ed70148 R09: 0000000000000000
[ 1107.361531] R10: 0000000000008000 R11: 0000000000000206 R12: 000055984ed5e860
[ 1107.361532] R13: 0000000020000000 R14: 00007ffb8e36d290 R15: 00007ffac864e6c0
[ 1107.361533] Code: 01 00 83 81 78 12 00 00 01 48 2b 05 ce 01 4d eb 48 c1 f8 06 48 c1 e0 0c 48 8d 04 d0 48 8b 56 10 48 03 05 c7 01 4d eb 48 83 ca 03 <48> 89 10 83 a9 78 12 00 00 01 5d c3 0f 1f 44 00 00 66 2e 0f 1f
[ 1107.361557] RIP: gen8_ppgtt_set_pde.isra.27+0x45/0x60 [i915] RSP: ffffbc75078778c8

xuguangxin, 2020-06-17:
@XinfengZhang , this is an important issue since it will hang the entire cpu stack. 
Please help check it.
thanks

XinfengZhang, 2020-06-17:
could we reproduce the issue by any stream?

xuguangxin, 2020-06-17:
yes, any 1080P stream. I can share stream with you if you need it

XinfengZhang, 2020-06-17:
is there i915_error_state?

HeJunyan, 2020-06-17:
It seems that the vaBeginPicture/vaEndpicture kind APIs are not MT safe. When only one thread call them, that's OK. But if multi-thread call them, this bug shows. 

Is Batch buffer operation/feeding MT safe?

xhaihao, 2020-06-17:
It works well for me on my ICL. You could try with a newer version of Linux kernel. I am using 5.4.0

HeJunyan, 2020-06-18:
We also hang on TGL.
Is all these APIs MT safe? When we delete all queues in the command line(one thread mode), then everything runs well.
I just notice that when get the command buffer,  the legacy way like
GpuContextSpecific::GetCommandBuffer will do
MOS_LockMutex(m_cmdBufPoolMutex) to protect.
While in Gen12 encoder's command buffer allocation, it uses global var, but I can not find the sync manner. 

Just a hint, maybe I am not correct.

xuguangxin, 2020-06-18:
I have updated my kernel to 5.4.0 on CFL.
8 channels can't easily reproduce the issue now.
But if I increase channel number to 12, it will hang the system at first run.

`gst-launch-1.0 -v tee name = t  filesrc location = ~/cin.264 ! h264parse ! vaapih264dec ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=320, height=240" ! t.  t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=320, height=240" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=640, height=480" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=1280, height=720" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=1920, height=1080" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=320, height=240" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=640, height=480" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=1280, height=720" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=1920, height=1080" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=320, height=240" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=640, height=480" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=1280, height=720" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink t. ! queue ! vaapipostproc ! "video/x-raw(memory:VASurface), format=NV12, width=1920, height=1080" ! queue max-size-buffers=5 ! vaapih264enc  ! fakesink `

xuguangxin, 2020-07-24:
@Xiaogang-Li are you able to reproduce my result?
thanks

XinfengZhang, 2020-08-16:
@xuguangxin @HeJunyan I could not reproduce the issue on SKL with latest driver, we already clean some global variable recently. could you help to check whether the issue already been fixed?

HeJunyan, 2020-08-17:
I also can not produce this on SKL at the first time. TGL has his issue when I checked.

Xiaogangli-intel, 2022-04-19:
We tried 12 and 20 channels on SKL/TGL, cannot reproduce this issue.
```
KMD: kernel 5.17.0-051700rc5-generic
media-driver/ gmmlib/libva: latest
```

Could you please try again with ENV. ?

xuguangxin, 2022-04-19:
@HeJunyan can you reproduce this on TGL?
thanks

Xiaogangli-intel, 2022-04-20:
> 

On TGL, could you add _i915.enable_guc=2_ in kernel parameters? That can ensure that HuC firmware can be loaded properly.

Xiaogangli-intel, 2022-04-24:
@HeJunyan how about the result of your tests?

HeJunyan, 2022-04-24:
I do not find it now on TGL.

Xiaogangli-intel, 2022-04-24:
do you mean you didn't reproduce this issue on TGL?

HeJunyan, 2022-04-24:
Yeah, I just tried times on TGL, but I can not produce it now.

Xiaogangli-intel, 2022-04-24:
Ok, thanks.
So this issue should be fixed in latest driver.
@xuguangxin if no other concern, I will close this issue.

closed 2022-05-05 by Xiaogangli-intel